OPCFoundation/UA-.NETStandard #2088: Is there a requirement that a client subscribe to a specific event on the server upon connecting?

htbmw, 2023-03-02:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

In the past I read somewhere that a client should create a specific subscription to something (most likely an event) on the server when connection is established.

UA Expert client creates this subscription and it can be seen when using the Alarms&Conditions sample server.

I searched online but am unable to find where I have read this information in the past.

My question is: Is there a specification that requires a client to create this specific subscription, or is it rather a best practice instead of a requirement?
Can someone supply me with a link if this information exists somewhere?

My guess is that this subscription could be something to allow the client session to be managed according to certain events on the server, for example a graceful shutdown when the server signals a scheduled shutdown or something?

We have issues where a session becomes stale after a certain idle period, and the client is unable to communicate with the server. This happens in kubernetes and not on a local computer. As part of our investigations, this subscription that I referred to above is something that I would like to investigate to determine if it is something that could assist in keeping a session active and alive at all times.

Any information / help / suggestions are most welcome!

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

ThomasNehring, 2023-03-03:
I think what you are looking for is described in Part 4 of the specification. If you fetch version 1.04, have a look at section 6.7. There you will see that the client should monitor the State variable in the ServerStatus (node id ns=0;i=2259), either by creating a subscription or by polling it, if the server does not support subscriptions. That section also explains how the client is supposed to detect connection issues and how to react on it. 

In the .Net stack Opc.Ua.Client project this is realized in the Session class. At the end of the ConnectAsync method a handler is assigned to the m_session.KeepAlive event handler, and the corresponding timer is set up in the StartKeepAliveTimer() method.

htbmw, 2023-03-06:
@ThomasNehring , thank you so much! That is exactly what I was looking for.
The specification is quite big and although I looked there, I must have missed it :)

closed 2023-03-06 by htbmw