OPCFoundation/UA-.NETStandard #2097: Bad  Too Many Subscriptions exeption

sauravsharma15, 2023-03-10:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

Hi

I am using the OPC Foundation.NetStandard.OPC.UA Version 1.4.370.12 . This is I am using to fetch the tags from OPCUA server . when We are trying to see the tags in UA Expert then its showing the data in view screen. But when we are trying to fetch the tag data in Code Application then its showing the error for some of the tags
Error Name : BadTooManySubscriptions. This error I am getting with the Tags which have little similar name and only different in the taxt with last 4 characters like example:

A01.Tags.Tag1
A01.Tags.Tag2
A01.Tags.Tag3

like above example the tags are present on certain OPCUA server. While fetching only the tags with little similar name are not able to add for the Subscription. Other Tags easily adding on Subscription.

Error is coming on subscription.Create();

When I have tried to analysis it then find out that its coming on the Tags having almost similar name or big length Name of Tag.

Can you Please give me detail what will be the reason for Error BadTooManySubscriptions.?

We are connect the OPCUA server using the TCP Connection through IP address.

Please guide what will be reason for the issue.

Thanks
Saurav Sharma

### Expected Behavior

BadTooManySubscriptions

### Steps To Reproduce

I am also getting this error of BadTooManySubscriptions. This error I am getting with the Tags which have little similar name and only different in the taxt with last 4 characters like example:

A01.Tags.Tag1
A01.Tags.Tag2
A01.Tags.Tag3

like above example the tags are present on certain OPCUA server. While fetching only the tags with little similar name are not able to add for the Subscription. Other Tags easily adding on Subscription.

Error is coming on subscription.Create();

### Environment

```markdown
- OS: Windows 10
- Environment:visual studio 2022
- Runtime:.NET  4.12.2.19454
- Nuget Version:1.4.370.12
- Component: Opc.Ua.Core
- Server: Cleint OPCua server
- Client: UA expert
```

mrsuciu, 2023-03-10:
@sauravsharma15 

The BadTooManySubscriptions service result is returned when the number of subscriptions on the Server has reached the maximum configured number of subscriptions. It's default value is 100 but it can be configured see  `<MaxSubscriptionCount> ` tag values from the reference server's xml configuration.
It seems that either you are creating too many subscriptions (when you probably do not need to) or your Server application needs to increase it's MaxSubscriptionCount value to support more.

sauravsharma15, 2023-03-13:
In  Opc.Ua.SampleClient.Config.xml I have tried to  mention <MaxSubscriptionCount>10000</MaxSubscriptionCount> But still getting same Issue.
I think this is may be due to almost same type of name  Like below

A01.Tags.Tag1
A01.Tags.Tag2
A01.Tags.Tag3

Please suggest How to resolve this issue as its creating problem to move forward with the testing of the Tag subscription for 1000 tags parallelly .

Thanks 
Saurav Sharma

MD-V, 2023-03-14:
@sauravsharma15 the problem is on server side. Which server are you trying to connect to?  
Possible solutions:
- Please disconnect from UAExpert before running your client. 
- Make sure that you only create one subscription on client side with multiple MonitoredItems.

EthanChangAED, 2023-06-20:
Closed the issue. Feel free to reactivate if the solutions don't work or there is any update.

closed 2023-06-20 by EthanChangAED