OPCFoundation/UA-.NETStandard #2099: Diagnostics Are Available For Anonymous Users

elliot-gawthrop, 2023-03-13:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The SDK makes some attempts to restrict diagnostics nodes for anonymous users, and the children of `SessionsDiagnosticsSummary` and `SubscriptionDiagnosticsArray` will not be accessible.

However the full diagnostics are still available by reading the Value of those nodes, rendering the above restrictions pointless.

### Expected Behavior

Expected that if the children of the diagnostics nodes be restricted, the value should also be restricted.

### Steps To Reproduce

1. Connect to server as anonymous
2. Read value of i=3707 and i=3708

### Environment

```markdown
Server: 1.4.370.12
```

elliot-gawthrop, 2023-03-13:
Related #1993

romanett, 2024-07-20:
Fix needs to be done here: https://github.com/OPCFoundation/UA-.NETStandard/blob/2aec78534756e4c044fae611818acadc56f2e36d/Libraries/Opc.Ua.Server/Diagnostics/DiagnosticsNodeManager.cs#L749

PR #2695 (2024-08-05): [Server] Limit Access to server diagnostics array to Admin User
## Proposed changes

There is already a function to limit access to some server diagnostics to only admin users, however the session diagnostics array was missed, this PR addresses this issue:

View with Admin:
[image]

View without Admin:

[image]

## Related Issues

- Fixes #2099 

## Types of changes

What types of changes does your code introduce?

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test [... cut]