OPCFoundation/UA-.NETStandard #2103: .NET7 Support

Flatscha, 2023-03-16:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

While upgrading to .NET7 I encountered a cycle loss on all my processes that use OPC.
Writes inconsistently use more time.

Normal:
2023-03-15 13:32:31.7260 | DEBUG | (Test1) Write Item: GVL_OPC.D2P.M1.IO [True], Time used **0,006101** 

Outliers:
2023-03-15 13:32:31.7260 | DEBUG | (Test1) Write Item: GVL_OPC.D2P.M1.IO [True], Time used **0,8563447** 
2023-03-15 10:56:52.6105 | DEBUG | (Test1) Write Item: GVL_OPC.D2P.M1.IO [True], Time used **1,0075487** 

The stopwatch only stops the _Session.Write_

### Expected Behavior

No outliers in writing speed

### Steps To Reproduce

Write a .NET7 console app that asynchronous writes onto variables on a OPC-server

Async only to get results quicker it works also in synchronous context

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.4.3
- Runtime: .NET 7
- Nuget Version: 1.4.371.60
- Component: Opc.Ua.Client
- Server: Beckhoff OPC
```

### Anything else?

Probably the problem is already solved just by adding .NET 7 support 
if not I do not have any clue why this problem happens

mregen, 2024-02-21:
.NET 8 is supported. Please re test and reopen if needed.

Flatscha, 2024-04-25:
Hi Martin,
upgraded to .NET 8 and Nuget Version 1.5.374.36 but unfortunately the problem remains.

mregen, 2024-06-19:
@Flatscha Please share the code snippet how you write and measure the outliers.

closed 2024-12-17 by mregen