OPCFoundation/UA-.NETStandard #2125: Call Method fail when method has only output parameter

LauraAmb, 2023-04-17:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I need to call a method without input arguments. When I call it, the response is 'BadNothingToDo'. I tried with two different servers and different methods and it doesn't work

### Expected Behavior

Return output arguments as UaExpert do.

### Environment

```markdown
- OS: Windows 10
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

MD-V, 2023-04-17:
@LauraAmb can you please show some example code?

LauraAmb, 2023-04-17:
The method is like the one in the image.
I found the node, but when I call the function IList<object> Call(NodeId objectId, NodeId methodId, params object[] args) of Session.cs, doesn't work. 
In this case inputArguments is empty. The function tests StatusCode and throws an exception (BadNothingTodo).

[image]