OPCFoundation/UA-.NETStandard #2127: <title>

cernydav, 2023-04-19:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [X] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Some Nuget packages of the project are licensed under MIT: 

OPCFoundation.NetStandard.Opc.Ua.Client
OPCFoundation.NetStandard.Opc.Ua.Server 

but rest as for example
OPCFoundation.NetStandard.Opc.Ua.Core 
OPCFoundation.NetStandard.Opc.Ua

are under more permissive OPC foundation license.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

### Anything else?

https://github.com/OPCFoundation/UA-.NETStandard/releases

mregen, 2023-04-19:
@cernydav please elaborate. Maybe this issue discussion helps: https://github.com/OPCFoundation/UA-.NETStandard/issues/1377

cernydav, 2023-04-20:
@mregen  Based on discussion in https://github.com/OPCFoundation/UA-.NETStandard/issues/1377  I assume that licenses are correctly aligned on nuget at this moment. So, we can rely on this information and this issue can be closed.

mregen, 2023-04-20:
great, thanks

closed 2023-04-20 by mregen