OPCFoundation/UA-.NETStandard #2136: X509Certificate2 major bug creates millions of private keys under C:\ProgramData\Microsoft\Crypto\Keys

eero-dev, 2023-04-24:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

This happens only when the application is using certificates, so when not using SecurityMode=None in OPC UA configuration

CheckApplicationInstanceCertificate creates2 private key files whenever CheckApplicationInstanceCertificate is called due to it calling new X509Certificate2(...)... other places in the code also call new X509Certificate2().

### Expected Behavior

Private key files are disposed and deleted from the disk by the library, since it calls the new X509Certificate2(...)

### Steps To Reproduce

1. Create a while loop here to simulate possible reconnection/app restart here: github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/Program.cs#L153
2. Observe C:\ProgramData\Microsoft\Crypto\Keys folder growing constantly

Full sample application based on ConsoleReferenceClient which recreates the bug can also be provided.

### Environment

```markdown
- OS: Windows 10, Windows 11, Windows Server 2016, Windows Server 2019
- Environment: VS 2022
- Runtime: NET 7.0
- Nuget Version: 1.4.371.86 
- Component: Client
- Server: KepServerEX v6 Simulator used for testing 
- Client: Custom based on ConsoleReferenceClient
```

### Anything else?

There has been previous investigations into this for example this blog demonstrates the behaviour and also explains the problem:
https://snede.net/the-most-dangerous-constructor-in-net/

eero-dev, 2023-04-29:
Upon further investigation, the issue seems to arise from the function 'LoadPrivateKey' in DirectoryCertificateStore which gets called by LoadPrivateKeyEx. It seems to be called once by LoadApplicationConfiguration and then by CheckApplicationInstanceCertificate. The offending code seems to be when enumerating the storageFlags, hopefully you can drop net462 support and default to using EphemeralKeySet flag.

For now we have default to using 'None' encryption mode with OPC UA, as the other modes cause disk flooding.

mregen, 2023-12-11:
Hi @eero-dev , I did some investigation with our standard codebase to reproduce the issue.
Because you mentioned net481 I was under the false impression this issue was opened for platforms which support bouncy castle. 
There is for now two minor improvements we can do for the enxt release
i) the legacy bouncy castle supported platforms always set the `PersistKey` flag, which means the key get even not deleted when a cert is disposed. I verified this flag can be removed.
ii) The support for `X509Store` on Windows requires the use of the `PersistKey` flag, but since newer code which uses the `CertificateRequest` API already deals with creating a copy of the key when added to the X509Store, on netstandard2.0 and net462 platforms the key can be deleted.
iii) The use of the `EphemeralKey` flag is in general possible, however when supporting the `opc.https` transport, kestrel or the https client refuse to use the certificate, so in this case the ephemeral keyset can not be used. There is some API extension necessary.

mregen, 2023-12-11:
@eero-dev for your sample code the solution could be quite simple. Only ever create a single instance of the client application and load the private key only once, even if you retry to connect.

eero-dev, 2023-12-11:
We have reduced the number to around 10 now, however we must keep our own logic around reconnection since the library has been way too buggy and we are unable to trust the reconnection that is built in. We are on NET 8. I assumed the use of PersistKey is because of the net462 support.

mregen, 2023-12-12:
Hi @eero-dev, I built a PR with the changes to support ephemeral keys. Almost all platform test still pass, due to the platform differences it may cause some regression, so taking it directly for the next release I consider high risk. 
If you can help to test it with your use case would help. I can also build a custom Nuget package if needed. 
Thanks, Martin

eero-dev, 2023-12-15:
It seems to be fixed and we are unable to see any negative side effects

PR #2421 (2023-12-22): Use EphemeralKeySet for certificates with private key on supporting platforms
## Proposed changes

- In order to reduce leaking private key files on windows, `EphemeralKeySet`  is used on supporting platforms.
- Keys from the Ephemeral Keyset can not be used with `CopyWithPrivateKey`, add support to copy these keys, e.g. for GDS push support.
- MacOS throws PlatformNotSupported when ephemeral keyset is used.
- Kestrel needs a dedicated key without the ephemeral keyset flag.
- Dispose private keys where possible to reduce number of leaked files in tests.
- For bouncy castle legacy support, do n [... cut]
closed 2023-12-22 by mregen