OPCFoundation/UA-.NETStandard #2137: Checking if self-signed client certificates are accepted

ValtteriL, 2023-04-25:
**Type of Issue**

Question x3

**Describe the Issue**

I am trying to check whether a server validates the client certificates it receives. My code:

```
using Opc.Ua;
using Opc.Ua.Client;
using Opc.Ua.Security.Certificates;

namespace Quickstarts.ConsoleReferenceClient
{
    public static class Program
    {
        public static async Task Main(string[] args)
        {
            Uri uri = new Uri("opc.tcp://echo:53530");

            Console.WriteLine("# Discovering applications");
            DiscoveryClient dc = DiscoveryClient.Create(uri);
            ApplicationDescriptionCollection adc = dc.FindServers(null);

            foreach (ApplicationDescription ad in adc)
            {
                foreach (string s in ad.DiscoveryUrls)
                {
                    // ask each discoveryUrl for endpoints
                    Console.WriteLine($"## Discovering endpoints for {ad.ApplicationName}:{s}");
                    DiscoveryClient sss = DiscoveryClient.Create(new Uri(s.Replace("opc.http", "http")));
                    EndpointDescriptionCollection edc = sss.GetEndpoints(null);

                    foreach (EndpointDescription ed in edc)
                    {
                        // try to start session with each endpoint
                        // ed.EndpointUrl = ed.EndpointUrl.Replace("opc.http", "http"); // <-- same exception with and without this
                        Console.WriteLine($"### Trying to start session with {ed.EndpointUrl}");

                        try
                        {
                            ConnectionUtil util = new ConnectionUtil();
                            var session = await util.StartSession(ed, new UserIdentity());
                            session.Close();
                            session.Dispose();

                            Console.WriteLine($"[!] {ed.EndpointUrl} accepts self-signed certificates!");
                        }
                        catch (Opc.Ua.ServiceResultException e)
                        {
                        
                            if (e.Message.Contains("Bad_SecurityChecksFailed") || e.Message.Contains("BadSecureChannelClosed"))
                            {
                                // nothing to do, the endpoint does not accept self-signed certificates
                            }
                            else
                            {
                                Console.WriteLine($"UNKNOWN EXCEPTION: {ed.EndpointUrl}");
                                throw;
                            }
                        }
                    }
                }
            }
        }
    }

    public class ConnectionUtil
    {
[... 95 lines cut ...]
UNKNOWN EXCEPTION: opc.https://echo:53443/OPCUA/SimulationServer
Unhandled exception. Opc.Ua.ServiceResultException: BadUnknownResponse
   at Opc.Ua.SessionClient.CreateSession(RequestHeader requestHeader, ApplicationDescription clientDescription, String serverUri, String endpointUrl, String sessionName, Byte[] clientNonce, Byte[] clientCertificate, Double requestedSessionTimeout, UInt32 maxResponseMessageSize, NodeId& sessionId, NodeId& authenticationToken, Double& revisedSessionTimeout, Byte[]& serverNonce, Byte[]& serverCertificate, EndpointDescriptionCollection& serverEndpoints, SignedSoftwareCertificateCollection& serverSoftwareCertificates, SignatureData& serverSignature, UInt32& maxRequestMessageSize)
   at Opc.Ua.Client.Session.Open(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain)
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ITransportWaitingConnection connection, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
   at Opc.Ua.Client.DefaultSessionFactory.CreateAsync(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
   at Quickstarts.ConsoleReferenceClient.ConnectionUtil.StartSession(EndpointDescription endpointDescription, UserIdentity userIdentity) in /home/valtteri/development/opcua-scanner/Program.cs:line 99
   at Quickstarts.ConsoleReferenceClient.Program.Main(String[] args) in /home/valtteri/development/opcua-scanner/Program.cs:line 34
   at Quickstarts.ConsoleReferenceClient.Program.<Main>(String[] args)
make: *** [Makefile:3: run] Error 134
```

Thus the HTTPS endpoints cause BadUnknownResponse exception.

The target server is the ProsysOPC Simulation Server.

The target server returns HTTPS endpoints with prefix "opc.https://". The library does not seem to understand that. I tried converting it to https:// in the EndpointUris, but I still get the same exception.

The library version I am using is: `1.4.371.60 @ 01/27/2023 12:33:12 -- 1.4.371.60+39078d6801adb84a51d9a552656d4cfe5dfc6161`

**Questions**

1. Why are the HTTPS endpoints discovered only when FQDN is used?
2. Is the exception a bug? If not, what am I doing wrong? How can I check whether the HTTPS endpoints check for client certificates or not?
3. The target server returns HTTPS endpoints with prefix "opc.https://". The library does not seem to understand that. Should it?

ValtteriL, 2023-04-26:
Thanks for the reply @mregen. Sad to hear about the lack of cycles.

_1. Why are the HTTPS endpoints discovered only when FQDN is used?_

I think you are right. The server does not recognize the FQDN and therefore will not take it into account and will return all endpoints. With only hostname it recognizes it, and takes the protocol into account as well and therefore only returns those starting with opc.tcp://.

_2. & 3._

I get the same `BadUnknownResponse` exception when calling any of the FindServers() here:
```
DiscoveryClient asd = DiscoveryClient.Create(new Uri("https://echo:53443"));

asd.FindServers(null);
asd.FindServers(null, "https://echo", null, null, out ApplicationDescriptionCollection adc);
asd.FindServers(null, "opc.https://echo:53443", null, null, out ApplicationDescriptionCollection adc);
asd.FindServers(null, "opc.https://echo", null, null, out ApplicationDescriptionCollection adc);
```
So I guess its a bug.

The lack of `opc.https://` -prefix support is more complex matter than just replacing it with `https://` everywhere. Is that correct?

mregen, 2024-01-18:
@ValtteriL the opc.https endpoints are supported since release 1.4.372. Please reopen if you find this still to be an issue.

closed 2024-01-18 by mregen