OPCFoundation/UA-.NETStandard #2144: Cannot browse nodes with quotationmarks in the node ID

MarisaGoergen, 2023-05-10:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

I'm trying to read nodes from a S7. The IDs of all PLC nodes contain Quotationmarks like `"Demo"."Static"."Test"`. If I try to browse all my nodes, I only see the ones without quotation marks. Has somebody got an idea? The UA Expert is totally fine. 

That is my used code, which should be quite similar to the example
```
 _session.Browse(null, null, nodeId, uint.MaxValue, BrowseDirection.Forward, ReferenceTypeIds.HierarchicalReferences, true, (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,
                    out _,
                    out var nextRefs);

foreach (var nextRd in nextRefs){
   ...
} 
```

The Description of the nodes contains "BadAttributeInvalid(0x80350000). Can that be the reason?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version: 1.4.371.60
- Component:
- Server: S7 1500
- Client: 1.4.371.60
```

mrsuciu, 2023-05-12:
@MarisaGoergen 

That shouldn't be the source of the problem. 
Does it still happen if you modify the ConsoleReferenceClient to browse the same target node ?

mregen, 2023-05-27:
Hi @MarisaGoergen , here is a sample to browse the whole address space: https://github.com/OPCFoundation/UA-.NETStandard/blob/d8f8a4e7ad007f6b131e01b7c0cbc74d0cd5dddf/Applications/ConsoleReferenceClient/ClientSamples.cs#L514
It has been tested with Siemens PLC and theses type of nodes can be browsed without an issue.