OPCFoundation/UA-.NETStandard #2151: Hostname in OPC UA Application’s certificates & config file can be in align with Kubernetes pod behavior?

charlesaugustineabb, 2023-05-17:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Hostname in OPC UA Application’s (Server and Client) config file is used in 3 places, which is used for creating certificate.
1)	<ApplicationUri>urn:localhost:UA:DataAccessServer</ApplicationUri>
2)	<SubjectName>CN=Quickstart DataAccess Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
3)	  
  <ServerConfiguration>
    <BaseAddresses>
      <ua:String>https://localhost:62547/Quickstarts/DataAccessServer</ua:String>
      <ua:String>opc.tcp://localhost:62548/Quickstarts/DataAccessServer</ua:String>
    </BaseAddresses>

PKi certificates Subject and Subject alternate name is based on above configuration, which is also validated in foundation SDK with  localhost, probably not in unified automation SDK case. 
In the case of Kubernetes, pod name and its ip can be keep changing during each deployment. However, in windows OS generally node name wont change. 

### Expected Behavior

OPC UA Application’s certificates can be in align with Kubernetes pod behavior?  or you could suggest me a solution if you have already?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-05-24:
Hi @charlesaugustineabb , you can leave out , `DC=localhost`. Then you can pick a name for the application URI. It may also a good way in a Kubernetes environment to switch to a no file configuration, e.g. this sample: https://github.com/OPCFoundation/UA-.NETStandard/blob/cea95d213a28da7e8507ea896f4b639b188a0f0c/Tests/Opc.Ua.Server.Tests/ServerFixture.cs#L57
to be able to use environment or command line values.
For the alternate names, its a bit tricky, ich you know the external hostname or IP you can specify it in the baseaddresses or alternatenames instead of localhost, they the pod host name is not being used. But in general its difficult if the IP addresses or hostnames keep  changing, the only choice is to ignore the hostname on the client side.

charlesaugustineabb, 2023-05-30:
Thanks for reply @mregen , i didnt got any mail notification when you answered .

as shown in image.
[image]

MyServer pod can go upto 3 replicas
since pod ip keep changing i will be using MyServerService with fixed ip which is ip5
in secret1 keeping the certficate , private key and trust list for MyServer Pod.
there will only one certficate which will be shared/used by MyServer pod's replica1 and replica2 and replica3.
ApplicationURI can be 
           urn:ip5:UA:MyServer  or urn:MyServerService:UA:MyServer

So Subject Alternative name value in certficate will like below

  URL=urn:ip5:UA:Quickstarts:MyServer
  DNS Name=ip5

  or
 
  URL=urn:MyServerService:UA:Quickstarts:MyServer
  DNS Name=MyServerService

Kindly let me know your valuable thoughts on above way to keep certificate in kubernet pod scenario ?

mregen, 2023-05-30:
Hi @charlesaugustineabb , this all looks good. To map a Kubernetes secret or ConfigMap may require some extra implementation. E.g. there is support to plug in custom certificate stores.

charlesaugustineabb, 2023-05-31:
Thanks again @mregen for sharing your views.

i have already done a  POC on creating secret , putting certificate and private key in secret and loading same by a Server (based on foundation sdk). sharing details below, 

**Creation of Secret from files.**

kubectl create secret generic cert-secret --from-file=test.pfx --from-file=test.der

**Yaml file : MyServer loading certficate and private key from secrets**

apiVersion: apps/v1
kind: Deployment
metadata:
  name: MyServer-deployment
  labels:
    app: MyServer
spec:
  replicas: 1
  selector:
    matchLabels:
      app: MyServer
  template:
    metadata:
      name: MyServerpod
      labels:
        app: MyServer
    spec:
      serviceAccountName: my-service-account
      containers:
      - name: MyServer
        image: abbiapcpdev.azurecr.io/MyServer:local2
        ports:
        - containerPort: 80
        volumeMounts:
        - name: gvr-volume-private
          mountPath: /app/MyServer/pki/own/private/
        - name: gvr-volume-cert
          mountPath: /app/MyServer/pki/own/certs/
      volumes:
        - name: gvr-volume-private
          secret:
            secretName: cert-secret
            items:
              - key: test.pfx
                path: test.pfx
        - name: gvr-volume-cert
          secret:
            secretName: cert-secret
            items:
               - key: test.der
                 path: test.der

charlesaugustineabb, 2023-05-31:
i think we can close the case

closed 2023-05-31 by charlesaugustineabb