OPCFoundation/UA-.NETStandard #2178: Auto-certificate generation appears to differ between version 1.4.371.86 and 1.4.371.91.

princenivre, 2023-06-07:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Our UA server calls application.CheckApplicationInstanceCertificate, which auto-generates a certificate.  When using the 1.4.371.86 UA library assemblies, the resulting certificate is generated with the provider, Microsoft Enhanced Cryptographic Provider v1.0.  When using version 1.4.371.91 assemblies, the certificate uses provider, Microsoft Software Key Storage Provider.

In the latter case, the certificate details cannot be viewed in the UA Configuration tool because cryptographic exception, "Invalid provider type specified" is thrown.  The cert from 86 works fine.  Of course, we'd like to use the latest stable release.  Any idea how to get version 91 to use the enhanced provider?

### Expected Behavior

Expect certificate created with 91 release to have PFX file using provider Microsoft Enhanced Cryptographic Provider v1.0, just like the 86 release.

### Steps To Reproduce

1. Build a UA server with 1.4.371.**86** libraries.
2. Ensure UA server certificates do not yet exist.
3. Cause UA server to invoke application.CheckApplicationInstanceCertificate(), such as by starting it up.
4. Use certutil.exe on the created PFX file.  Note the provider is Microsoft Enhanced Cryptographic Provider v1.0.
5. Build a UA server with 1.4.371.**91** libraries.
2. Ensure UA server certificates do not yet exist.
3. Cause UA server to invoke application.CheckApplicationInstanceCertificate(), such as by starting it up.
4. Use certutil.exe on the created PFX file.  Note the provider is Microsoft Software Key Storage Provider.

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: .NET 4.8
- Nuget Version:
- Component: Opc.Ua.Security.Certificates
- Server: Custom
- Client:
```

princenivre, 2023-06-07:
I found the issue.  It's not the release version causing the issue, it is the .NET framework targeted.  net4.6.2 behavior differs from net4.8 (I think).  I'd still like to know why the cert with enhanced provider seems to work better than those with the key storage provider?

mregen, 2023-06-09:
Hi @princenivre, .NET 4.6.2 uses bouncy castle to create the certificate, .NET 4.8 has already support for the built in .NET CertificateRequest API. The latter is the preferred way to create certificates due to the dependency to third party crypto libraries which should be avoided. The .pfx is also generated by bouncy castle in .NET 4.6.2 and .NET export by 4.8. Again the latter is the preferred way.  Would be interesting to know what kind of issue the UA Configuration tool has. Maybe it also needs a recompile  with a newer .NET version?

mregen, 2023-08-29:
please reopen if there are still open questions

closed 2023-08-29 by mregen