OPCFoundation/UA-.NETStandard #2189: ApplicationInstance deletes certificates and keys out of the X509 Store

chrisstankevitz, 2023-06-15:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When `Opc.Ua.Configuration.ApplicationInstance` determines that the application certificate is "invalid", then the class invokes `DeleteApplicationInstanceCertificate` which deletes the certificate.

At https://github.com/OPCFoundation/UA-.NETStandard/blob/1.4.357.28/SampleApplications/SDK/Opc.Ua.Configuration/ApplicationInstance.cs#L1008:

When `!certificateValid`, this routine calls `CreateApplicationInstanceCertificate`.

If the application certificate and key are stored in the X509Store, then the routine **overwrites** the application's certificate and key.

The actual delete happens via a call to `DeleteApplicationInstanceCertificate` at https://github.com/OPCFoundation/UA-.NETStandard/blob/1.4.357.28/SampleApplications/SDK/Opc.Ua.Configuration/ApplicationInstance.cs#L1259

I'm specifying my application certificate using a thumbprint as shown below:

```
    <ApplicationCertificate>
      <StoreType>X509Store</StoreType>
      <StorePath>LocalMachine\My</StorePath>
      <Thumbprint>12ab980a4493a22ad02712ab980a4412ab980a44</Thumbprint>
    </ApplicationCertificate>
```

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-06-15:
Hi @chrisstankevitz , I guess the expectation is that the certificate is never deleted if the thumbprint is specified, right?

chrisstankevitz, 2023-06-15:
@mregen thank you.  FYI I believe the cert/key pairs were also being deleted when a) loaded based on the CN (not just the fingerprint) and b) when using the Directory/filesystem store.

Yes: expectation is (was) that the stack would not delete my CA-signed certs.

Doesn't seem to be happening on master.  We will start the process of testing with the new version, but in the meantime our production system is vulnerable.  Looking at the code, I don't see an obvious way to work around this "delete if invalid" but please let me know if you have any ideas.

You can close the issue -- thank again.

chrisstankevitz, 2023-06-15:
This is a duplicate of #1102

closed 2023-06-15 by mregen