OPCFoundation/UA-.NETStandard #2190: Question about Certificates

Blazej1981, 2023-06-15:
I'm new in OPC UA protocol and i'm trying to understand how the certificates works in code.
I saw that the examples use a .exe file called "Opc.Ua.CertificateGenerator.exe" for generate the certificates but the question is, there is a way for create a certificate through code without use the "Opc.Ua.CertificateGenerator.exe"?
I've another question about certificates in examples code.
I had run the "ConsoleReferenceServer" application, obviously has been created a new certificate but how can i use an existing certificate for the application instead of create a new one?
About the private keys...is the file .pfx created in "OPC Foundation\pki\own\private" my private key of the certificate that will be created? If no...how is created the .pfx file in "OPC Foundation\pki\own\private"? Where is my PC private key?

mrsuciu, 2023-06-16:
@Blazej1981 

You can follow the call to Opc.Ua.CertificateFactory.CreateCertificate(string subjectName) which is available also in the [OPCFoundation.NetStandard.Opc.Ua.Core nuget](https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua.Core).

It can be referenced in a separate utility console application of your choice as follows:

```
X509Certificate2 certificate = CertificateFactory.CreateCertificate(
              "CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost")
              .SetLifeTime(24)
              .SetRSAKeySize(2048)
              .CreateForRSA();
```
Here the value of the subjectName is picked out from the following XML tag belonging to Quickstarts.ReferenceServer.Config.xml:

 `<SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>`

You will have to write out the .der (public certificate) and .pfx (private key) files out of the certificate  but you can check how this is done in the code and adapt to your needs.

If you already have an existing valid application certificate (public cert in .der format under the "certs" folder and private key in .pfx format under the "private" folder) at the same certificate store location (the one specified under the  `<ApplicationCertificate><StorePath>[PATH]</StorePath></ApplicationCertificate>` in the xml configuration file), than the application will pick it up.

Yes the private key of the application certificate is contained in the file mentioned by you with the .pfx extension.

Blazej1981, 2023-06-16:
> @Blazej1981
> 
> You can follow the call to Opc.Ua.CertificateFactory.CreateCertificate(string subjectName) which is available also in the [OPCFoundation.NetStandard.Opc.Ua.Core nuget](https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua.Core).
> 
> It can be referenced in a separate utility console application of your choice as follows:
> 
> ```
>             X509Certificate2 certificate = CertificateFactory.CreateCertificate(
>                "urn:localhost:UA:Quickstarts:ReferenceServer",
>                "Quickstart Reference Server",
>                "CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost",
>                new List<string> { Utils.GetHostName() })
>                .SetLifeTime(24)
>                .SetRSAKeySize(2048)
>                .CreateForRSA();
> ```
> 
> Here the value of the subjectName is picked out from the following XML tag belonging to Quickstarts.ReferenceServer.Config.xml:
> 
> `<SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>`
> 
> You will have to write out the .der (public certificate) and .pfx (private key) files out of the certificate but you can check how this is done in the code and adapt to your needs.
> 
> If you already have an existing valid application certificate (public cert in .der format under the "certs" folder and private key in .pfx format under the "private" folder) at the same certificate store location (the one specified under the `<ApplicationCertificate><StorePath>[PATH]</StorePath></ApplicationCertificate>` in the xml configuration file), than the application will pick it up.
> 
> Yes the private key of the application certificate is contained in the file mentioned by you with the .pfx extension.

Thank you!

closed 2023-06-16 by Blazej1981