OPCFoundation/UA-.NETStandard #2192: Opc Tcp not responding for 30sec when send the malformed packets to the port

utthungagokul, 2023-06-16:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Hi Team,

 We have currently faced a behavior of OPC UA Server not responding for more than 30sec with malformed packets.
 This was Identified from our customer location where they check for the Security testing and found the issue.
Basically, security team was trying to send the malformed packets to OPC TCP port with more than 200 bytes over the network by incrementing based each socket connection. 

Looking forward to the solution.

Thanks 
Gokulkanna Balusamy

### Environment

```markdown
- OS: Windows 10
- Environment: 
- Runtime: .Net Core 3.1
- Nuget Version:
- Component: OPCFoundation.NetStandard.Opc.Ua
- Server: 1.4.370.12
- Client:
```

mregen, 2023-06-18:
Hi @utthungagokul , please report security issues as described here: https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SECURITY.md

with instructions or sample code how to repro.

utthungagokul, 2023-07-13:
Issue has reported to Security team, so closing the issue here.

closed 2023-07-13 by utthungagokul