OPCFoundation/UA-.NETStandard #2196: By implementing FileTransfer in server, the FileDirectoryState objects have too much children

GhislainPr, 2023-06-20:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Hello,
By implementing FileTranser in a server, I create FileDirectoryState objects.
From 1.04.371 version, the created objets have 2 unexpected children named <FileName> and <FileDirectoryName>.

### Expected Behavior

The FileDirectoryState objects may not have <FileName> or <FileDirectoryName> children.

### Steps To Reproduce

In a NodeManager derived class, you override CreateAddressSpace method to do that :
-  var _fileSystem = new FileDirectoryState(this, parent);
-  _fileSystem.Create(SystemContext, null, browseName, null, true);

you will get <FileName> and <FileDirectoryName> children objects by calling _fileSystem.GetChildren(...)

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mrsuciu, 2023-06-21:
@GhislainPr 

This behavior is a bug related to the presence of optional place holders in the instance of the objects and will be fixed in the soon to be released maintenance nugget package. Stay tuned.

mregen, 2023-06-27:
@GhislainPr please check if the issue is fixed with 1.4.371.96.

EthanChangAED, 2023-07-25:
Closing this issue. @GhislainPr feel free to re-activate if you can still reproduce this issue with the latest version.

closed 2023-07-25 by EthanChangAED