OPCFoundation/UA-.NETStandard #2203: Add async delegate handlers when address is called

juaal12, 2023-06-23:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Right now when an address is called the delegate handler works synchronously. An async function is not able to be used since it is not supported by the package. Just a small workaround has to be done but it is not the best way of handling Tasks in my opinion. Find here some attached code:

```
public ServiceResult Handle(ISystemContext context,
    IList<object> inputArguments, IList<object> outputArguments)
{
    // Start the asynchronous operation on a separate thread
    var task = Task.Run(() => HandleAsync(context, inputArguments, outputArguments));

    // Synchronously wait for the task to complete
    task.Wait();

    // Return the result of the completed task
    return task.Result;
}

private async Task<ServiceResult> HandleAsync(ISystemContext context,
    IList<object> inputArguments, IList<object> outputArguments)
{
    // Your asynchronous logic here
}
```

### Expected Behavior

I think the package should handle async methods. Not sure it is supported by the communication standard but if it is supported, the async delegate handler should be implemented in this package.

### Steps To Reproduce

1. Open reference server project
2. Go to reference node manager and for Add function for example, try to attach to the handler an async method.
3. It wont work so you will have to use the workaround written on this comment.

### Environment

```markdown
The environment is not important here.
```

mregen, 2023-06-27:
Hi @juaal12, is this for server development? Client has async service calls.

juaal12, 2023-06-27:
> Hi @juaal12, is this for server development? Client has async service calls.

Yes this is for server development. Right now when you create an address space and you call it using an external tool for example UA Expert, if the handler is synchronous it works like a charm but if the handler is asynchronous it does not get called.

crega, 2025-02-05:
Any news regarding this?
I also have similar issue. When I am binding `MethodSet.OnCallMethod` since it's not async delegate I have to do similar workaround with `Task.Run(...`

If I would try to implement Task based approach for method execution, what else would be necessary to change as well ?

romanett, 2025-02-06:
@crega if your delegates are called from a nodeset it is most likely code generated by the ModelCompiler. So the ModelCompiler would need to support async delegates. I am not shure if this is the case:
https://github.com/OPCFoundation/UA-ModelCompiler

closed 2025-08-12 by marcschier