OPCFoundation/UA-.NETStandard #2207: CTT 1.4.410 errors on latest release

EthanChangAED, 2023-06-29:


mregen, 2023-07-20:
the following Mantis issues have been opened:

https://mantis.opcfoundation.org/view.php?id=9033
https://mantis.opcfoundation.org/view.php?id=9034
https://mantis.opcfoundation.org/view.php?id=9035
https://mantis.opcfoundation.org/view.php?id=9036
https://mantis.opcfoundation.org/view.php?id=9037

mregen, 2023-07-20:
The following issues or missing tests were discovered:

CTT Missing tests (1.4.9.410 2023)
	- Implement event trigger nodes for CTT
	- Implement HA nodes for tests
	- Implement YArrayItem / XYArrayItem
	- Implement Image (e.g.BMP) Datatypes
	- Fix test cases for monitored items
	- Implement Node management services
	- Only allow connection of sha1 signed certs with Basic128Rsa15

mregen, 2023-07-20:
@mrsuciu did you find other tests / cases missing?

mrsuciu, 2023-07-21:
No

EthanChangAED, 2023-08-10:
New items are created to breakdown the works needed.

[CTT] Implement event trigger nodes for CTT
--
[CTT] Implement HA nodes for tests
[CTT] Implement YArrayItem / XYArrayItem
[CTT] Implement Image (e.g.BMP) Datatypes
[CTT] Fix test cases for monitored items
[CTT] Implement Node management services
[CTT] Only allow connection of sha1 signed certs with Basic128Rsa15

closed 2023-08-10 by EthanChangAED