OPCFoundation/UA-.NETStandard #2221: Reference UAClient reconnect failure

DvdKhl, 2023-07-11:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I'm using the [UAClient](https://github.com/OPCFoundation/UA-.NETStandard/blob/043a56c3f2edb0b1117b64a9eea34e341de51aff/Applications/ConsoleReferenceClient/UAClient.cs) class to connect to an industrial device which I do not have control over.  
At some unknown point the connection is lost and fails to reconnect.

The sequence is mostly the same:
- After the successful connection has been in use for a while (hours or days) the server seems to stop to respond to keep alive requests
- After around 2min the log message "BadRequestTimeout" is continually writtenevery ~2s
- After ~15min the log messages changes to "BadSecureChannelClosed"
- After ~2s the log messages changes again to "BadSessionIdInvalid" from which it never recovers
- Only after creating a new UAClient instance is this issue resolved until the connection fails again

### Expected Behavior

Automatically reconnect without having to create a new Session manually.

### Steps To Reproduce

Cannot reproduce reliably
1. Open Program.cs in sample and replace connection/node details
2. Compile and run Sample
3. Wait for Successful connection and DataEvents
4. Turn Opc Server Device off and wait for error messages
5. Turn Opc Server Device on again
6. Wait for successful reconnect

[OPCReconnectFailure.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/12011161/OPCReconnectFailure.zip)

### Environment

```markdown
- OS: Debian
- Environment: Visual Studio 2022 17.6.4
- Runtime: .NET6
- Nuget Version: 1.4.371.91
- Component: Opc.Ua.Client
- Server: Industrial Device (Unknown Opc Server Software)
- Client: Reference UAClient
```

### Anything else?

Full Logs:
[OpcReconnectFailure.csv](https://github.com/OPCFoundation/UA-.NETStandard/files/12011135/OpcReconnectFailure.csv)

mregen, 2023-07-18:
thanks @DvdKhl for sharing the bits, not obvious whats going on, need to take a deeper look.

raulpsj, 2023-08-08:
Hi @DvdKhl,

Try version 1.4.367.100, I think it's the same problem #1796

galigula2, 2024-02-05:
Hi, what it the status of this? And are there viable workarounds e.g. re-creating the UAClient when this starts to happen?

I'm seeing something similar with build against .NET 6 and running in a Linux machine.