OPCFoundation/UA-.NETStandard #2259: <title>

svetovid, 2023-08-07:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When the latest LTS version (6.0.x) of `Microsoft.Extensions.Logging.Abstractions` is used together with `OPCFoundation.NetStandard.Opc.Ua.Core` in a net48 project,
error "Could not load file or assembly 'Microsoft.Extensions.Logging.Abstractions, Version=3.1.23.0, Culture=neutral, PublicKeyToken=adb9793829ddae60" occurs.

### Expected Behavior

Microsoft.Extensions.Logging.Abstractions supports .NET Framework 4.6.1 and all .NET 4.8 projects can easily reference`Microsoft.Extensions.Logging.Abstractions 6.0.4`, so should `OPCFoundation.NetStandard.Opc.Ua.Core` .
`OPCFoundation.NetStandard.Opc.Ua.Core` works with other assemblies which have LTS version in .NET Framework 4.8 projects 

### Steps To Reproduce

1. create a .NET Framework 4.8 project
2. add reference to OPCFoundation.NetStandard.Opc.Ua.Client (including OPCFoundation.NetStandard.Opc.Ua.Core)
3. update Microsoft.Extensions.Logging.Abstractions to use version 6.0.4
4. invoke Session.Create(...) of OPCFoundation.NetStandard.Opc.Ua.Client package
5. error occurred

### Environment

```markdown
- OS: Win10
- Environment: VS2022
- Runtime:.NET 4.8
- Nuget Version: 1.4.371.41
- Component: OPCFoundation.NetStandard.Opc.Ua.Client
- Server:
- Client:
```

### Anything else?

This is because OPCFoundation.NetStandard.Opc.Ua.Core requires 3.x version in .NET Framework 4.8, due to the following build condition in https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Stack/Opc.Ua.Core/Opc.Ua.Core.csproj
```
<Choose>
<!-- Note: Due to incompatibilities of Microsoft.Extensions Nuget packages between versions 3.x and 7.0,
         use latest versions only on .NET 5/6, otherwise 3.1.x -->
<When Condition="'$(TargetFramework)' == 'net5.0' OR '$(TargetFramework)' == 'net6.0'">
<ItemGroup>
<PackageReference Include="Microsoft.Extensions.Logging.Abstractions" Version="7.0.1" />
</ItemGroup>
</When>
<Otherwise>
<ItemGroup>
<PackageReference Include="Microsoft.Extensions.Logging.Abstractions" Version="3.1.32" />
</ItemGroup>
</Otherwise>
</Choose>
```

JensNordenbro, 2023-08-07:
Avoiding upgrading is also hard since other nugets are requiring higher versions.

mregen, 2023-08-12:
Hi @svetovid, in the past there were warnings issued in the ci pipeline and in other projects when using latest libraries with .NET framework, but I don't see this anymore here: #2233 
Is it preferable to use 6.0.4 or 7.0.1 with .NET Framework 4.8? 
.NET framework support for 4.6.2 has been deprecated in the .372 release.

closed 2023-08-14 by mregen

JensNordenbro, 2023-08-14:
It is a good question but since 7-versions include net4-series assemblies these are the latest so I would assume that is the correct version: 
[image]

Any reason for choosing 6-series for net48 @mregen ?

JensNordenbro, 2023-08-14:
I see now that @svetovid requested version 6.

mregen, 2023-08-14:
version 6 is LTS, please let me know if there is any problem upgrading even to 7.x with this preview Nuget package: https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua.Core/1.4.372.46-preview

svetovid, 2023-08-17:
@mregen thank you very much!
Now it works with https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua.Core/1.4.372.46-preview.
Will there be the same issue, once dotnet 8 has been released?