OPCFoundation/UA-.NETStandard #2282: NodeStateFactory instance creation behavior

symarchand, 2023-08-25:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

In the context of trying to implement AddNodes method, when calling the NodeStateFactory.Create method to create an instance of an object, the create NodeState does not seem to comply with OPC UA specification part 4 on [AddNodes](https://reference.opcfoundation.org/Core/Part4/v104/docs/5.7.2).

The created instance does not create its mandatory children.

### Expected Behavior

I would expect the node to be instantiated and initialized according to UA specification of AddNodes.

### Steps To Reproduce

1. Populate the type list of the NodeStateFactory
2. Create an instance using the NodeStateFactory

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: dotnet6.0
- Nuget Version: 1.4.371.96
- Component: Opc.Ua.Core
- Server: Custom server based on StandardServer SDK class
```

### Anything else?

It seems that Initialize methods generated by the model compiler are not called when creating an instance with NodeStateFactory.
It seems also that they are all protected, except the ones that need an already initialized object.

By playing with Reflection to force the call of Initialize (for test purpose only of course :-)), I managed to see an apparently complete object on a client. Unfortunately, by going into details, I saw that mandatory children were not really present (BadNodeIdUnknown) and optional ones were pointing to the object type's corresponding object or variable. The test was done creating a LimitAlarm object.

I am not sure if I am doing it the intended way. If not, what would that way be ?
If I do it the right way, I am not sure whether the problem comes from the NodeStateFactory, or the model compiler, or both.

EthanChangAED, 2024-03-22:
@symarchand , thank you so much for reporting this issue. To double check, are you still seeing this issue?

If yes, would you share a little more about the repro step? (e.g. implementation, or the scenario) It looks like a common scenario. However, we didn't hit this issue with our basic validation. So I am suspecting there should be some more specific factors to trigger this problem.

symarchand, 2024-03-25:
Hello @EthanChangAED, 
I will try to reproduce the issue and post more details, probably next week.

symarchand, 2024-04-17:
Hello, 
I reproduced the issue using sample applications in version 1.4.372.0-preview
To do so, I added a method in the BoilerNodeManager to create a LimitAlarm. I call this method once the server is started.

```
public void AddLimitAlarm()
{
    NodeHandle handle = Server.NodeManager.GetManagerHandle(Opc.Ua.ObjectIds.ObjectsFolder, out _) as NodeHandle;

    AddNodesResult result = new AddNodesResult();
    AddNodesItem node = new AddNodesItem {
        BrowseName = new QualifiedName("myAlarm"),
        ParentNodeId = handle.NodeId,
        NodeClass = NodeClass.Object,
        ReferenceTypeId = Opc.Ua.ReferenceTypeIds.HasComponent,
        RequestedNewNodeId = new NodeId("myAlarm", m_namespaceIndex),
        TypeDefinition = Opc.Ua.ObjectTypeIds.LimitAlarmType
    };

    AddNode(null, node, result);
}

public virtual ServiceResult AddNode(OperationContext p_context, AddNodesItem p_nodeToAdd, AddNodesResult p_result)
{
    if (p_nodeToAdd == null) throw new ArgumentNullException(nameof(p_nodeToAdd));
    if (p_nodeToAdd.RequestedNewNodeId == null) throw new ArgumentNullException(nameof(p_nodeToAdd.RequestedNewNodeId));
    if (p_result == null) throw new ArgumentNullException(nameof(p_result));

    p_result.StatusCode = StatusCodes.Bad;

    // Check Browse Name
    if (p_nodeToAdd.BrowseName == null)
    {
        return StatusCodes.BadBrowseNameInvalid;
    }

    // Get Parent Node State
    var parent = GetNodeState(p_nodeToAdd.ParentNodeId, out var parentNodeManager);
    if (parent == null)
    {
        return StatusCodes.BadParentNodeIdInvalid;
    }

    // Creating node state
    var nodeState = CreateNodeState(p_nodeToAdd);
    if (nodeState == null)
    {
        return StatusCodes.BadNodeClassInvalid;
    }

    nodeState.BrowseName = p_nodeToAdd.BrowseName;
    nodeState.DisplayName = p_nodeToAdd.BrowseName.Name;
    nodeState.SymbolicName = p_nodeToAdd.BrowseName.Name;

    // Get Node Id
    var nodeId = new NodeId(p_nodeToAdd.RequestedNewNodeId.Identifier, p_nodeToAdd.RequestedNewNodeId.NamespaceIndex);

    // Checking if requested node id is already assigned
    if (Server.NodeManager.GetManagerHandle(nodeId, out _) != null)
    {
        return StatusCodes.BadNodeIdExists;
    }

    nodeState.NodeId = nodeId;

    // Adding attributes
    nodeState = FillAttributes(p_nodeToAdd, nodeState, null, out var resultAttributes);
    if (StatusCode.IsNotGood(resultAttributes))
    {
        return resultAttributes;
[... 183 lines cut ...]
    var objectStateTypes = assembly.GetTypes().Where(t => t.IsSubclassOf(typeof(BaseInstanceState)));

    foreach (var typeType in objectStateTypes)
    {
        string instanceStatetypeName = typeType.Name;

        NodeId nodeId = null;
        var idsType = typeof(Opc.Ua.ObjectTypeIds);
        var idMembers = idsType.GetMember(typeType.Name.Replace("State", "Type"));//.Substring(0, typeType.Name.Length - "State".Length));
        if (idMembers.Length == 0) continue;
        var idMember = idMembers[0];

        if (idMember.MemberType == MemberTypes.Field)
            nodeId = ((FieldInfo)idMember).GetValue(null) as NodeId;
        else if (idMember.MemberType == MemberTypes.Property)
            nodeId = new NodeId((uint)((PropertyInfo)idMember).GetValue(null));

        Server.DefaultSystemContext.NodeStateFactory.RegisterType(nodeId, typeType);
    }
}
```
Doing so, I get the following result on my client: 
[image]
Which seems fine, except that optional child nodes point to nodes defined in the Type of the object, and mandatory nodes returns BadNodeIdUnknown when read.
[image]

romanett, 2026-02-14:
@marcschier  Isnt this fixed with the new Source generator?