OPCFoundation/UA-.NETStandard #2284: Memory leak in TcpMessageSocketAsyncEventArgs

andrerodi, 2023-08-26:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When calling the 'ReadValue' or the Async method from the 'Session' class, this will create a memory leak due to 'TcpMessageSocketAsyncEventArgs' not being disposed of properly.

This behavior also happens when a 'MonitoredItem' fires its event notifications.

I run the OpcUaClient as a singleton in a 'BackgroundService'.

Every bump in the following picter is one call of 'ReadValueAsync'
[image]

I saw that in inside the 'UaSCUaBinaryChannel' class is a protected 'WriteOperation' class that implements `ChannelAsyncOperation<T> : IAsyncResult, IDisposable`

```
        public void Dispose()
        {
            Dispose(true);
            GC.SuppressFinalize(this);
        }
```

GC is suppressed. Is this intended?

After reading the value, im using Mapsters 'Adapt<T>' to map from PLC structs (ExtensionObject) to my DTO classes. Atomic datataypes will get their value with the 'GetValue<T>' method from the DataValue object.

Maybe I'm being a moron and I'm doing something wrong here?

### Expected Behavior

Objects being disposed of propperly.

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.7.2
- Runtime: .NET Core 7.0.5
- Nuget Version: Client - 1.4.372.56 | ComplexTypes - 1.4.372.56
- Component: Opc.Ua.Client
- Server: Siemens S7 1500
```

### Anything else?

Perfview diff of before vs after: [image]

andrerodi, 2023-08-26:
I forgot to mention. To somehow mitigate this problem I call

```
GC.Collect();
GC.WaitForPendingFinalizers();
```

After every read/notification.
It's costly in CPU cycles, but that's the only way that pops into my head for a quick and dirty fix(?)

mregen, 2023-08-27:
@andrerodi thanks for the report, I will try to reproduce the issue.

andrerodi, 2023-08-27:
@mregen Thanks. If you have a Siemens PLC plus TIA (I use V17) laying around, I could create a minimal project for you.

mregen, 2023-08-28:
Hi @andrerodi, I have TIA V17 + S7 PLC, so the project might be helpful if the root cause is a special complex type or configuration.

I have been testing with the console client and S7 with command line as follows: `opc.tcp://<ipaddress>:4840 -b -s -c -a` which doesn't expose a memory leak. Once the subscription is counts of active memory and objects stay put. Could you check if you can repro with the console app? 

Other finding, the dispose of the Write is never called, so the GC.SuppressFinalize is not active. The TcpMessageSocketAsyncEventArgs may not get disposed for a different reason (e.g. specific codepath).

andrerodi, 2023-08-28:
> @andrerodi one more question, is this a regression since .371?

It has been a problem ever since I started using this package. I first installed the nuget around April/Mail of 2022. Don't know which version was the latest.

This problem only really gets noticeable when I read from big data structures that are deeply nested (nested level of 2 or more and with arrays of 100 items etc.). 
I will try to get a minimal project running by this weekend and will post again asap with a link to the repo.

Thanks in advance @mregen for your help.

mregen, 2023-08-28:
Hi @andrerodi, if you read really large nested structures, for subscriptions there are a few caches that can be turned of to improve the situation. E.g. `Subscription.DisableMonitoredItemCache = true` and `Subscription.MaxMessageCount = 1`.
For the complex type system, try to clear the NodeCache after loading the complex types:
`ISession.NodeCache.Clear()`.

andrerodi, 2023-08-28:
@mregen I added a `FastDataChangeSubscribeAsync<T>(string node, Func<MonitoredItem, T, Task> handleAsync)` method to make use of the FastDataChangeCallback to my code, changed the `Subscription` settings and cleared the `NodeCache` like you described, but to no avail.

Still mitigating this problem by calling

```
GC.Collect();
GC.WaitForPendingFinalizers();
```

at the very end of the `FastDataChangeHandler` method I implemented.
I will setup a minimal C# and TIA project so that you can replicate this. I'll let you know ASAP.

andrerodi, 2024-01-24:
@mregen Sorry for the vast delay. I was in an unforseeable and elongated crunch time. In the end, it was a GC-Collector setting. It was set to server GC mode, but needed client GC mode.

closed 2024-01-24 by andrerodi