OPCFoundation/UA-.NETStandard #2288: Question: TranslateBrowsePathsToNodeIds behavior in corner cases

okapies, 2023-08-29:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [X] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I want to know how `TranslateBrowsePathsToNodeIds` service works in the following request:

- `BrowsePath(startingNode=<NodeId: /Objects>, relativePath=null)`
- `BrowsePath(startingNode=<NodeId: /Objects>, relativePath=<RelativePath: elements=[])`
- `BrowsePath(startingNode=<NodeId: /Objects>, relativePath=<RelativePath: elements=[<Element: targetName=null>])`
- `BrowsePath(startingNode=<NodeId: /Objects>, relativePath=<RelativePath: elements=[<Element: targetName="">])`

I also wonder how "" (empty) and "/" as [RelativePath text format](https://reference.opcfoundation.org/Core/Part4/v104/docs/A.2) should be treated in the client/server implementation.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

okapies, 2023-09-01:
The [spec](https://reference.opcfoundation.org/Core/Part4/v104/docs/7.26) says:

> The final element may have an empty targetName. In this situation all targets of the references identified by the referenceTypeId are the targets of the RelativePath.

In this situation, what is the "empty targetName"?

- targetName = null
- targetName = '' (empty string)
- targetName = QualifiedName(name=null, namespaceIndex=[index])
- targetName = QualifiedName(name='', namespaceIndex=[index])

okapies, 2023-09-01:
If path string `/[idx]:` is given, `RelativePathFormatter` returns `QualifiedName(name=null, namespaceIndex=0)` only for `ns=0` and otherwise `QualifiedName(name="", namespaceIndex=[idx])`. Which behavior is correct?

https://github.com/OPCFoundation/UA-.NETStandard/blob/c7381cf9f7a4577184ee5a09035ea46e3db74ea0/Stack/Opc.Ua.Core/Types/Utils/RelativePath.cs#L844-L850

EthanChangAED, 2023-11-30:
migrating to discussion board.