OPCFoundation/UA-.NETStandard #2291: <title>

premfort, 2023-08-29:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

The OPC UA client application reports the error "Certificate chain validation incomplete" on creating Session and fails to establish the connection with the OPC UA server. 

Here is the complete exception captured in the OPC UA client application log.
`Error details: Opc.Ua.ServiceResultException: Certificate chain validation incomplete. ---> Opc.Ua.ServiceResultException: Certificate chain validation incomplete.
   --- End of inner exception stack trace ---
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain, ConfiguredEndpoint endpoint)
   at Opc.Ua.Client.Session.Open(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain)
   at Opc.Ua.Client.Session.<Create>d__84.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)`

Configuration:
- OPCFoundation.NetStandard.Opc.Ua version: 1.4.370.12
- Certificates Own, Issuer, Trusted and Rejected are configured in client application code and the store type is directory. 
- AutoAcceptUntrustedCertificates is true
- Security Mode: Sign, Security Policy Uri: http://opcfoundation.org/UA/SecurityPolicy#Basic256, Encoding Support: Required
- Provided valid user identity
- DisableDomainCheck was false

Performed the below changes and tested. These did not help and getting the same error.
1). Manually copied the certificate from rejected folder to trusted folder.
2). Created Issuer and CRL directory.
3). Manually copied the certificate to Issuer directory.
4). Set the SendCertificateChain to true
5). Updated the latest OPCFoundation.NetStandard.Opc.Ua version 1.4.372.56 based on the [issue fix](https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FOPCFoundation%2FUA-.NETStandard%2Fissues%2F2020&data=05%7C01%7CPrem.Kannan%40Honeywell.com%7C89fc87199582400ce69408dba87af3ce%7C96ece5269c7d48b08daf8b93c90a5d18%7C0%7C0%7C638289018268688995%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=SKwmKeses78SLGvR%2FMNJ9TR0EqEh9oFEA4jEAqJrPNg%3D&reserved=0)

It  was reproducible with another OPC UA client which uses the OPCFoundation.NetStandard.Opc.Ua version 1.4.371.60

### Expected Behavior

OPC UA client should validate the certificate and establish the connection with OPC UA server.

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.6.6 
- Runtime: NETStandard.Library 2.0.3
- Nuget Version: 1.4.370.12
- Component:
- Server: 
- Client:
```

EthanChangAED, 2023-09-05:
@premfort, thank you for the feedback. We have not observed this issue recently in the reference server/client. Have you ever tried the later Nuget (**1.4.732.96)** version and compare the reference with your application?
- https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ReferenceClient
- https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ReferenceServer

EthanChangAED, 2023-09-11:
Close as no further information was provided. Feel free to reopen if there are updates.

closed 2023-09-11 by EthanChangAED

Siemensplcuser, 2025-07-30:
Even I face this issue when I connect to the built-in OPC UA Server of Siemens TIA PLC. Cannot really figure out if the issue is with the NuGet package or the PLC.

romanett, 2025-07-30:
@Siemensplcuser please provide more details