OPCFoundation/UA-.NETStandard #2292: Data Delivery Delay

gurc4n, 2023-08-31:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

I have an application in production.
The production depends on the application. If there is a delay in the response of the application, it is a micro stoppage.
I have not updated the Nuget package lately, I updated it to the latest version in the middle of 7th month.
The production started to report a lot of delays. I store the ServerTimeStamp in the database, when I check the application's log files, there is a delay in the delivery of the tag change. It is completely random, I cannot find a reason. There is no reconnection or publish error or something like that.
As an example; the ServerTimeStamp: 2023-08-30 22:40:10.5094352, and the method triggered at 2023-08-30 22:40:14.540. Delay in milliseconds: 4031

I have reverted the version of the application.
Starting this morning the library version is **1.4.371.50 and all the delays are gone**.
There is no difference in the logic or something like that, most probably, I had to adopt the ReconnectHandler but the rest of the application stays the same.

The only evidence I can find is whenever that delay happens, I also get a TCP error from the Redis library, it says "No connection available".

I am out of ideas. Could you please help me?

### Environment

```markdown
- OS: Windows Server 2022
- Environment: Visual Studio 2022
- Runtime: .NET 4.8, .NET Standard
- Nuget Version: latest/1.4.371.50
- Component: Opc.Ua.Client
- Server: KepwareEX
- Client: Almost the clone of the reference client.
```

mregen, 2023-09-01:
Hi @gurc4n , the latest version is 1.4.372.56, or did you update to 1.4.371.96 which shows the stoppages?

gurc4n, 2023-09-04:
Hi @mregen 
1.4.371.96 --> This is the version I had problems. I have waited to answer the question, wanted to see the results again, sorry for the delay.
For 4 days I have no delays. (Actually, I have only 1 out of 411029, I think it is ignorable)

EthanChangAED, 2023-09-05:
@gurc4n, thanks so much for sharing the updates. It seems good now, but feel free to let us know if you see more delays in the future.

gurc4n, 2023-09-05:
@EthanChangAED , 
So, it is fixed in the latest version?
Did I catch it correctly?
Maybe I was not clear, Everything is good on version "1.4.371.50", it was bad on "1.4.371.96". 

As a temporary workaround, the application is using the old version of the library. 
Should I upgrade to the "**current**" latest? Do you think the issue is solved in the latest version?

EthanChangAED, 2023-09-05:
@gurc4n, I think I miss-understood your comment.

Is the summary below correct?
- 1.4.371.50 - For 4 days I have no delays. (Actually, I have only 1 out of 411029
- 1.4.371.96 - Encountered lot of delays

mrsuciu, 2023-09-05:
@gurc4n As I understand you are using a Redis server to save the data change notifications and you've observed that "No connection available" errors are reported by the client Redis library, which point to the possibility that the Redis server connection is not available at that time.
Since you report that there is a correlation between the Redis library error message and the delayed notifications (which not necessarily implies causation), can you rule out that this is not an issue induced by the client Redis library itself while trying to store the data received in the data change callback ?
Are you performing the write to the database synchronously in the FastDataChangeCallback and can the write take more time to complete when such errors occur ?

Can you reproduce this behavior with the latest version ?

GurcanBicer, 2023-11-09:
I'm sorry for not getting back to you sooner. I had many other problems. I stayed at 1.4.371.50 for a while, and everything was perfect.
I updated the package to the latest version, 1.4.372.76, at the beginning of the week.

I had a new exception occasionally, BadRequestInterrupted. The **delays** were there too...

I cannot say this was because of a network problem, as I have the same application in four plants, and all four plants had the same problem.

All plants also ended up with a Publish Error / Reconnect loop.

There is nothing special about what I am doing using the library. I simply use the sample application to implement the connect/reconnect/subscription/read/write. In straightforward terms, the application does the following: a tag changes, I catch the change on that tag, read a bunch of other tags in that notification, and write the results in the database.

I create a session at the beginning of the application, and I create a subscription for each production line. That means approximately ten subscriptions, each with 20-30 monitored items.

I stopped using FastDataChangeCallback, but to answer your question, @mrsuciu, no, I didn't do a sync database write in FastDataChangeCallback.

GurcanBicer, 2023-11-16:
The message timed out in the backlog attempting to send because no connection became available, command=GET, timeout: 5000, outbound: 0KiB, inbound: 0KiB, inst: 75, qu: 120, qs: 0, aw: True, bw: CheckingForTimeoutComplete, rs: ReadAsync, ws: Idle, in: 0, in-pipe: 0, out-pipe: 0, last-in: 0, cur-in: 0, sync-ops: 542171, async-ops: 16287, serverEndpoint: 10.5.176.31:61472, conn-sec: 20437.81, aoc: 0, mc: 1/1/0, mgr: 10 of 10 available, clientName: EPDI-APP014, IOCP: (Busy=1,Free=999,Min=50,Max=1000), WORKER: (Busy=59,Free=32708,Min=50,Max=32767), v: 2.7.4.20928 (Please take a look at this article for some common client-side issues that can cause timeouts: https://stackexchange.github.io/StackExchange.Redis/Timeouts)

This is the error message from Redis, and the OPC.UA delays also correspond.
I have increased the MinWorkerThread to 50; the busy count is 59. I am having some thread starvation!