OPCFoundation/UA-.NETStandard #2302: Error using the UA package in Unity - PlatformNotSupportedException: Operation is not supported on this platform

DiogoFran, 2023-09-08:
I constructed a client OPC UA from this library, in Visual Studio tool, after that I compiled the app and published it, to create dlls files. After that, I copy dlls files and paste into plugins folder, in the Unity project. Create a .cs file on Unity and copy lines of code from my first project (Visual Studio) into this file. 
When I run the code in Unity I get this error: "PlatformNotSupportedException: Operation is not supported on this platform.
Opc.Ua.Security.Certificates.CertificateBuilder.CreateForRSA () (at <7f3e3a168a4645dd83943693f0edea89>:0)
Opc.Ua.Configuration.ApplicationInstance.CreateApplicationInstanceCertificate (Opc.Ua.ApplicationConfiguration configuration, System.UInt16 keySize, System.UInt16 lifeTimeInMonths) (at <a10f441f612f4b578e3511a691a57205>:0)
Opc.Ua.Configuration.ApplicationInstance.CheckApplicationInstanceCertificate (System.Boolean silent, System.UInt16 minimumKeySize, System.UInt16 lifeTimeInMonths) (at <a10f441f612f4b578e3511a691a57205>:0)". 
So, I think, this method is not supported by the Android platform (which is my target platform).

Other notes:
- API compatibility level (Unity): .Net Standard 2.1
- Scripting backend (Unity): IL2CPP
- Unity version: 2022.1.22
- SO version: Windows
- Platform Target(Unity): Android

What can I do? All tips are appreciated. Thank you!! :)

EthanChangAED, 2023-09-11:
Hello DiogoFran,
Thanks for your feedback. If I understand your approach correctly, I think this library does not support your setup. Please refer to the "How to build ..." readme:
- https://github.com/OPCFoundation/UA-.NETStandard#how-to-build-and-run-the-reference-samples-in-visual-studio-on-windows 
- https://github.com/OPCFoundation/UA-.NETStandard#how-to-build-and-run-the-console-samples-on-windows-linux-and-ios

DiogoFran, 2023-09-11:
Thank you for your reply. 
I think so, i can use the library without security policies (certificates) but the client opc ua will have security problems.

closed 2023-09-12 by EthanChangAED