OPCFoundation/UA-.NETStandard #2323: Sending read request with higher number of nodes throws exception

LarsStegman, 2023-09-28:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When sending a request to read more than `MaxArrayLength` nodes to the server an exception will be thrown. 

```
Opc.Ua.ServiceResultException : Could not send request to server.
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location ---
   at Opc.Ua.SessionClient.ReadAsync(RequestHeader requestHeader, Double maxAge, TimestampsToReturn timestampsToReturn, ReadValueIdCollection nodesToRead, CancellationToken ct)
   at Opc.Ua.SessionClientBatched.ReadAsync(RequestHeader requestHeader, Double maxAge, TimestampsToReturn timestampsToReturn, ReadValueIdCollection nodesToRead, CancellationToken ct)
   at <MY CODE FROM HERE>
```

The library does make a check whether OperationLimits are not breached, but it does not check before hand whether channel limits will be breached.

### Expected Behavior

The library automatically batches requests when the channel limits will be breached.

### Environment

```markdown
- OS: Windows 10
- Environment: Rider
- Runtime: .NET 6
- Nuget Version:
- Component: Opc.Ua.Client
- Server: Umati Sample Server
- Client: This library
```

EthanChangAED, 2023-12-06:
@LarsStegman , any update from your side based on the previous question?

LarsStegman, 2023-12-07:
I'm sorry, I completely forgot to respond. I hope I have some time next week to create something.