OPCFoundation/UA-.NETStandard #2325: Calling CreateCACertificateAsync() in CertificateGroup of GDS (Global Discovery Server) does not place the certificate in TrustedListPath Certificate Store

romanett, 2023-10-02:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I am calling the Method CreateCACertificateAsync() on the "Default" CertificateGroup being created in the GlobalDiscoverySampleServer.

I am calling this method to replace the CA Certificate with a new one.

As you can see the new CA Certificate (with Name IOP-2023 CA) is only placed in the Authorities CerticateStore, but not in the "trustedList"-Store. Clients are pulling their CRLs from this store, so they don´t get the updated crls ...

[image]

I am not shure, why this behaviour occurs, as this copying is normally triggerd by the method UpdateAuthorityCertInTrustedList()

### Expected Behavior

New created CA is placed also in the TrustedList-CertificateStore

### Steps To Reproduce

1. Start GDS
2. Call Method UpdateAuthorityCertInTrustedList()
(I will improve this description tomorrow)

### Environment

```markdown
- OS: Ubuntu / Docker / mcr.microsoft.com/dotnet/aspnet:7.0
- Environment: Visual Studio 2022
- Runtime: .NET 7
- Nuget Version:
- Component: OPC.Ua.Gds.Server.Common
- Server: GlobalDiscoverySampleServer
- Client: -
```

### Anything else?

Here you can find the respository of the used application:
https://github.com/romanett/GDSwithREST

This has a rest endpoint on https://localhost:8081/swagger

There you can call the endpoint UpdateCA to trigger the method  UpdateAuthorityCertInTrustedList()

romanett, 2023-10-09:
Found the cause of the Issue
This happens when you change the SubjectName to a different one than the one set in the CertificateGroup-Constructor.

@mregen Should i change the method to not take the parameter (as the method should just use the fixed SubjectName), or should I create a Setter for the Subject Name?

As the CreateCACertificateAsync() method is public, this behaviour is in my opinion a Bug

mregen, 2023-10-10:
@romanett thanks for the feedback, nice work to build a REST interface for the GDS management. I hope to try it out some time. 
To your question I think it depends on the use case how a CA certificate should be updated - with or without the Subject name change. Since the configuration is saved as xml it may be better to not allow to change the subject name. Or in this case update the xml and restart the complete server when that happens.

PR #2336 (2023-11-17): GDS: add ArgumentException to Method CreateCACertificateAsync
## Proposed changes

Introduces an exception, to notify the user when supplying an invalid argument to the method CreateCACertificateAsync
This exception was introduced, as in the current implementation the same subject name has to be used as in the configuration or else the certificate is not copied to the trusted store.

## Related Issues

- Fixes #2325

## Types of changes

What types of changes does your code introduce?

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds fun [... cut]
closed 2023-11-17 by mregen