OPCFoundation/UA-.NETStandard #2326: <title>

htbmw, 2023-10-03:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Would it make sense to add support for BigInteger to the TypeInfo class?

My application receives a value from a JSON message that it then writes to an OPC UA node.
The value is deserialized as a BigInteger when the value is a large number like UInt64.MaxValue.

I already use TypeInfo.Cast for all other values that can be received via the JSON message, but BigInteger currently cannot be converted and the cast returns null.

using Opc.Ua;
using System.Numerics;

Console.WriteLine("Converting BigInteger....");

var bigInteger = new BigInteger(UInt64.MaxValue);

var uint64Integer = TypeInfo.Cast(bigInteger, BuiltInType.UInt64);

Console.WriteLine($"Converted: {uint64Integer}");

Actual Output:
===========
Converting BigInteger....
Converted: 

### Expected Behavior

Expected Output:
============
Converting BigInteger....
Converted:  18446744073709551615

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-10-26:
like the idea...

mregen, 2023-11-02:
Hi @htbmw, a BigInteger can be smaller / bigger than a Int64/UInt64. How do you envision the cast when the overflow condition is hit?

htbmw, 2023-11-02:
Hi @mregen 

To be honest I haven't thought about this beyond using UInt64.MaxValue/MinValue.
In my test I assert that the Min and MaxValue for a node of a particular numeric type can be handled by the json.
OPC UA won't allow a value outside the range of that type to be written to the node. Also it won't give you a value outside that range on a read.
Therefor I didn't think beyond the Min and Max ranges.

I think an overflow exception should suffice?

mregen, 2023-11-14:
Hi @htbmw, the Overflow exception would work, could you provide the implementation with a few test cases?

closed 2025-12-03 by romanett