OPCFoundation/UA-.NETStandard #2328: GetBuiltInType for abstract numeric types Number, Integer and UInteger return null

htbmw, 2023-10-03:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

When calling TypeInfo.GetBuiltInType() for a node that has an abstract numeric data type like Number, Integer, UInteger, the value returned is null.
I had a look in the TypeInfo code and I can see the logic that returns null:

`if (builtInType > BuiltInType.DiagnosticInfo && builtInType != BuiltInType.Enumeration)
{
    return BuiltInType.Null;
}`

But I do not understand why.
At the moment I am trying to write a value to a node that has one of the abstract numeric datatypes.
In order to do this, I need to convert my value accordingly.
I cannot do this without knowing if I am dealing with a Number, Integer or UInteger.

Any help, suggestions and/or explanations will be appreciated.

`var builtInType = TypeInfo.GetBuiltInType(new NodeId(28));

Console.WriteLine(builtInType);`

Actual Output:
Null

### Expected Behavior

TypeInfo.GetBuiltInType() should return the corresponding BuitInType for Number, Integer and UInteger data types.

`var builtInType = TypeInfo.GetBuiltInType(new NodeId(28));

Console.WriteLine(builtInType);`

Expected Output:
UInteger

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-10-26:
Hi @htbmw, Number, Integer and UInteger are only internally in the UA stack represented as BuiltInTypes, and they are abstract. Returning the Null is intentional. 

You can check the nodeId also like this:
 
```
NodeId nodeId = DataTypeIds.UInteger;
if (nodeId == DataTypeIds.UInteger)
{
    var valueUInteger = new Variant((UInt16)10);
    // write value etc.
}
```

mregen, 2023-11-14:
please reopen if this issue is not answered

closed 2023-11-14 by mregen