OPCFoundation/UA-.NETStandard #2342: Timeout Error when Reading Nodes after 12 Hours of Inactivity

caradilin, 2023-10-13:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

I am encountering an issue with an OPC UA client-server connection. After establishing a connection and setting the session timeout to 60000, I am experiencing a timeout error when attempting to read nodes after 12 hours of inactivity. The error message states: "System.ArgumentException: Timeout must be greater than zero. Parameter name: timeout." This indicates that the timeout value is either zero or an invalid value.

### Expected Behavior

No mistakes should be made

### Environment

```markdown
- OS: win10
- Environment:
- Runtime:
- Nuget Version:1.4.372.56
- Component:
- Server: 
- Client: 1.4.372.56
```

caradilin, 2023-11-14:
[2023 -11-13.Opc.Ua.log](https://github.com/OPCFoundation/UA-.NETStandard/files/13345294/2023.-11-13.Opc.Ua.log)

mregen, 2023-11-16:
Hi @caradilin , this is a very interesting log. 

1) any idea why the logger is restarted over and over again? In general it is best to use the implementation with an external `SetLogger` based on the `ILogger`. Not sure if this is done already.

2) I noticed that the keep alive of the subscriptions is very long.
e.g. `11/13/2023 16:54:42.445 For subscription 61, Keep alive count was revised from 4294967295 to 1431655765`
Keep alive count should be in the 5 or 10 second range, to keep the subscription alive. Similar for the session lifetime.

3) Second there is maybe a race condition in the implementation of the reconnect. 
I see sometimes multiple keep alive registrations and reconnect attempts.
Please follow the latest guidance of implementation of the reconnect and keep alive in the UAClient sample here: https://github.com/OPCFoundation/UA-.NETStandard/blob/37252f31b2dd729384ff69ba71c825324fb0bcaf/Applications/ConsoleReferenceClient/UAClient.cs#L321C15-L321C15
Specifically the session must only be replaced in reconnect complete if it is a new one, only then the old one should be disposed.

Please let me know if that leads to improvements.

caradilin, 2023-11-17:
> Hi @caradilin , this is a very interesting log.
> 
> 1. any idea why the logger is restarted over and over again? In general it is best to use the implementation with an external `SetLogger` based on the `ILogger`. Not sure if this is done already.

> A：This is because there are currently multiple OPC UA servers, with a persistent connection established to the server with port number 8888, and a on-demand connectivity status with the server on port 8855. The logging output does not differentiate between the connections to the different servers. I will make the necessary distinction in the future.

> 2. I noticed that the keep alive of the subscriptions is very long.
>    e.g. `11/13/2023 16:54:42.445 For subscription 61, Keep alive count was revised from 4294967295 to 1431655765`
>    Keep alive [...]

> A：Yes, for the persistent connection, I have set uint.MaxValue for the KeepAliveCount of subscriptions. However, for the session, I am using the default value for the KeepAliveInterval.

```c#
Subscription m_subscription = new Subscription(m_session.DefaultSubscription)
            {
                PublishingEnabled = true,
                PublishingInterval = 0,
                KeepAliveCount = uint.MaxValue,
                LifetimeCount = uint.MaxValue,
                MaxNotificationsPerPublish = uint.MaxValue,
                Priority = 100,
                DisplayName = key
            };
```
The question you just asked is exactly what I wanted to ask: When I set the session’s Timeout to 60000, but the subscription has a LifetimeCount of uint.MaxValue and there is an active connection between the client and the server, does it mean that the client’s session will never expire?
If subscriptions stay active, it is still not guaranteed that the client's session will never expire. Besides changing the session's timeout, is it possible to use a "watchdog" mechanism with periodic read and write requests to keep the session active?

> 3. There is maybe a race condition in the implementation of the reconnect.
>    I see sometimes multiple keep alive registrations and reconnect attempts.
>    Please follow the latest guidance of implementation of the reconnect and keep alive in the UAClient sample here: https://github.com/OPCFoundation/UA-.NETStandard/blob/37252f31b2dd729384ff69ba71c825324fb0bcaf/Applications/ConsoleReferenceClient/UAClient.cs#L321C15-L321C15
>    Specifically the session must only be replaced in reconnect complete if it is a new one, only then the old one should be disposed.

> A：Thank you for your suggestion, I will make the necessary changes and verify again. Afterward, I will share the project with you.

> Please let me know if that leads to improvements.

mregen, 2023-11-18:
```c#
Subscription m_subscription = new Subscription(m_session.DefaultSubscription)
            {
                PublishingEnabled = true,
                PublishingInterval = 0,  // this means fastest and then it is determined by the server, rather set a value in ms, e.g. 1000
                KeepAliveCount = uint.MaxValue,  --> set to multiple of publishing interval , e.g. 5 = 5 seconds
                LifetimeCount = uint.MaxValue, --> set to multiple of publishing interval , e.g. 600 = 10 minutes
                MaxNotificationsPerPublish = uint.MaxValue, 
                Priority = 100,
                DisplayName = key
            };
```

> The question you just asked is exactly what I wanted to ask: When I set the session’s Timeout to 60000, but the subscription has a LifetimeCount of uint.MaxValue and there is an active connection between the client and the server, does it mean that the client’s session will never expire?

The server can revise the subscription lifetime count. E.g. if a server doesn't support the TransferSubscription service, it doesn't make sense to allow longer lifetime for the subscription than the session. If it does, a very long subscription lifetime may lead to keep subscriptions alive even when the client has disconnected, for a long time. It should only be set so if it is intended. 
Even then it should be limited to a reasonable time in which events/data changes can be kept in server queues, e.g. 10 minutes.

> If subscriptions stay active, it is still not guaranteed that the client's session will never expire. Besides changing the session's timeout, is it possible to use a "watchdog" mechanism with periodic read and write requests to keep the session active?
> 
such a keep alive mechanism is implemented in the client, for both subscription and session. Only the session one is triggered by the client, for the subscription it is handled by the server by responding with keep alive publish responses.

EthanChangAED, 2023-11-30:
closing this issue. feel free to reopen if there is any udpate.

closed 2023-11-30 by EthanChangAED