OPCFoundation/UA-.NETStandard #2347: the project licence

kimdiego2098, 2023-10-20:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

My project nuget references opc.ua.client. What protocol do I need to comply with? I read the open source license instructions and it seems to be GPL, does it mean that my project needs to be open source

mregen, 2023-10-20:
Hi @kimdiego2098 ,here are the terms: http://opcfoundation.github.io/UA-.NETStandard/help/license_agreement.htm

kimdiego2098, 2023-10-20:
thinks @mregen  I saw this terms, but I still have some doubts. Does this mean that UA.NET cannot be used for private libraries for free?

mregen, 2023-10-21:
Hi @kimdiego2098, you can use the core library for free under the GPLv2 license, other libraries use the MIT license

mregen, 2023-10-26:
please reopen if necessary

WBMPH, 2023-11-23:
Sorry to re-open, but the license referenced in the NuGet packages specify they're licensed under MIT.

https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua.Client/1.4.372.76

[image]

I need some clarification.

kimdiego2098, 2023-11-23:
@WBMPH 

This does have ambiguity. The ua.core project is licensed under the gpl license, but other references such as ua.client are licensed under the mit license. so my conclusion is that I must follow the gpl2.0 license in order to use this open source library for free

mregen, 2023-11-28:
Hi @WBMPH and @kimdiego2098, 
since the client library uses core, due to the dependency the GPLv2 license is to be accepted for your project if you are not a corporate OPC Foundation member.
The license of MIT for the other libraries reflects source code contributions made to these libraries. 
According to the headers, these libraries only contain MIT licensed files, thats why the library is also flagged as MIT licensed.

WBMPH, 2023-11-28:
Please update the nuget package to reflect the correct licensing model.

mregen, 2023-11-29:
@WBMPH the packages are all up to date.

closed 2023-12-05 by EthanChangAED

JpWebster, 2025-09-24:
Glad I'm not the only one confused with the MIT, RCL, GPL licensing. If the Client Library consumes Core and core is GPL then shouldn't the client also be under RCL/GPL? 

This is very problematic for applications that use the Client library package.

JensNordenbro, 2025-09-24:
That is what @mregen wrote I think?! . In essence advertised as MIT but application becomes GPL transitively.  Like a modern Trojan horse🤪

JpWebster, 2025-09-25:
@JensNordenbro @mregen - Is there a member or contact at the OPC Foundation that can add clarity around the licensing of these components? In my understanding the OPCFoundation.NetStandard.Opc.Ua.Client (MIT) package is a derivative work from OPCFoundation.NetStandard.Opc.Ua.Core (GPLv2)? 

"If your plugin or package links to a GPLv2 DLL, it is considered a combined work. The entire combination (plugin + GPLv2 DLL) must be GPLv2. If your main application loads that plugin, it may also be subject to GPLv2"

Thanks.

mregen, 2025-09-27:
>  Is there a member or contact at the OPC Foundation that can add clarity around the licensing of these components?

Adding @opcfoundation-org and @marcschier