OPCFoundation/UA-.NETStandard #2355: How can I write a byte[] using this client?

KircMax, 2023-10-26:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

When I read a byte[] from a server (e.g. S7-1500) it will work but the Variant inside the DataValue will have the typeInfo ByteString which is logical since the c'tor for the variant with byte[] sets the m_typeInfo to TypeInfo.Scalars.ByteString ... 
This is also very comfortable for ByteString handling.
But what I can't get to work is now writing this ByteArray back to the server since it'll be encoded like a ByteString and therefore the server won't accept the value provided.

### Expected Behavior

It is possible to write the value that I got from the Read.

### Steps To Reproduce

1. Create a server with an array of byte that is readable and writable
2. Connect to the server with the client to read that variable
3. Try to write that variable with the value you got from 2.

### Environment

```markdown
- OS: Windows 10
- Environment: 
- Runtime:
- Nuget Version: 1.4.366 but I could see the same Variant implementation in the current repository
- Component: 
- Server: S7-1500 V3.0
- Client: .NETStandard
```

### Anything else?

I cant attach the wireshark recording so I'll attach screenshots.
[image]
[image]
[image]
[image]
[image]

KircMax, 2023-10-26:
I can/should probably also try it out with the current implementation, I just saw that the variant implementation for byte[] is still the same.

mregen, 2023-10-26:
Hi @KircMax , I tried this with a S7 as well, but I received a Byte array just fine, maybe it has to do how the array is defined, the one I tested has fixed length. But imho that might be a decoding bug.

To initialize the Variant as byte array e.g. 
```
var typeInfo = new TypeInfo(BuiltInType.Byte, 1);
var variant = new Variant(new byte[10] { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9 }, typeInfo);
```
But if a value is read from the server to write it back a fresh DataValue is necessary, e.g.:
```
NodeId readNode = "ns=3;s=\"Scalar.Simulation.Arrays\".\"Byte\"";
var value = await uaClient.Session.ReadValueAsync(readNode).ConfigureAwait(false);
// write back
var writeValue = new DataValue(value.WrappedValue));

// Write the node
WriteValueCollection nodesToWrite = new WriteValueCollection();

WriteValue writeVal = new WriteValue();
writeVal.NodeId = readNode ;
writeVal.AttributeId = Attributes.Value;
writeVal.Value = writeValue ;
nodesToWrite.Add(writeVal);

StatusCodeCollection results = null;
DiagnosticInfoCollection diagnosticInfos;

// Call Write Service
session.Write( null, nodesToWrite, out results, out diagnosticInfos);
```
Hope this helps

KircMax, 2023-10-27:
Hi @mregen,
I re-tried with the latest version and there it is also fine for me. 
Seems like it was fixed - sorry I bothered you.
[image]
Thanks a lot for your help, initializing it 'by hand' might be a workaround for now.

closed 2023-10-27 by KircMax