OPCFoundation/UA-.NETStandard #2361: OPC UA Server with https-uabinary behind a reverseproxy with different port

BoBiene, 2023-10-31:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When the port of AlternateBaseAddress is different from the Https-Endpoint port the discovery returns no valid endpoints.

See: https://github.com/OPCFoundation/UA-.NETStandard/blob/71ad9afd1cc46117887dbcec27d01d68d79d3a36/Stack/Opc.Ua.Core/Stack/Server/ServerBase.cs#L1155

### Expected Behavior

The port should not be required to be the same on the opc ua application and on the reverse proxy

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2023-11-02:
Hi Bo, could you check if [this](https://github.com/OPCFoundation/UA-.NETStandard/pull/1687) change caused a regression? 

I noticed also on the latest IOP that in some combination when discovery is done with AlternatBaseAdresses no endpoint was returned. This case should also be fixed.

mregen, 2024-01-18:
i think the problem generically applies when an alternate address is exposed as a different port, e.g. behind a NAT or when running in a docker container.

mregen, 2024-02-07:
Unfortunately the fix caused another regression, we need to take another look.

PR #2362 (2024-01-19): Fix GetEndpoints when https-uabinary is behind reverse proxy with different port
Skip port validation for https endpoints

## Proposed changes

Describe the changes here to communicate to the maintainers why we should accept this pull request. If it fixes a bug or resolves a feature request, be sure to link to that issue.

## Related Issues

- Fixes #2361

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-bre [... cut]
PR #2511 (2024-02-17): Ensure endpoints are returned when alternate url port does not match
## Proposed changes

- Ensure endpoints are returned even if the client url port do not match (similar to client)
- Improve endpoint tests

## Related Issues

- Fixes #2361 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [x] Test enhancement (non-breaking change to increase test coverage)
- [ ] [... cut]
closed 2024-02-17 by mregen