OPCFoundation/UA-.NETStandard #2365: Connect to an endpoint without discover

ReIaxo, 2023-11-03:
Hi, I want to connect to an  OPCUA Server without executing a endpoint discovery. I have the URL (i.e. 192.168.2.1:4840) and want to call session.create(...) but for this I need a ConfiguredEndpoint object. Is there a way to construct it manually? 
Or is there another way to do this?

I want this because I think the Server I need to connect to bisbehaves a little when I call DiscoveryClient.GetEndpoints(..)

Or would you say this is a no go?

Thank you in advance.

epasinetti, 2023-11-04:
I think the big problem of this library is lack of documentation and examples so complex, and mixed with a lot of stuff not related to the scope ... (utilities for logging, for manage parameteres and whataver ... ) 
The library is full of functionalities but not well explained ... your request is just one of many wishes.

mregen, 2023-11-06:
Hi @Relaxo and @epasinetti , feedback received. 
It is possible to create the configured endpoint manually, but it requires the server certificate to compare the returned server certificate with the certificate used in the endpoint configuration.
Also do not set the `updateBeforeConnect` parameter as it triggers a discovery as well.