OPCFoundation/UA-.NETStandard #2373: Trouble with "Session.Create"

NewMabo, 2023-11-09:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Hi,
I have trouble with "Session.Create".
I use a code in the nuget version 1.4.371.96 it's work.
But when I update the nuget to a new version (1.4.372.56 or 1.4.372.76) my session is never create.
My code stay in "Session.Create()" and nothing happened. 

Code:
public async Task Connect(EndpointDescription endpointDescription, bool userAuth, string userName, string password)
{
    //Secify application configuration
    Opc.Ua.ApplicationConfiguration ApplicationConfig = mApplicationConfig;

    //Hook up a validator function for a CertificateValidation event
    ApplicationConfig.CertificateValidator.CertificateValidation += Notification_CertificateValidation;

    //Create EndPoint configuration
    EndpointConfiguration EndpointConfiguration = EndpointConfiguration.Create(ApplicationConfig);

    //Connect to server and get endpoints
    ConfiguredEndpoint mEndpoint = new(null, endpointDescription, EndpointConfiguration);

    //Creat a session name
    string sessionName = "SessionOPCUAClientNET";

    //Create user identity
    UserIdentity UserIdentity = userAuth ? new UserIdentity(userName, password) : new UserIdentity();

    //Update certificate store before connection attempt
    await ApplicationConfig.CertificateValidator.Update(ApplicationConfig);

    //Create and connect session
    Session = await Session.Create(
        ApplicationConfig,
        mEndpoint,
        true,
        sessionName,
        60000,
        UserIdentity,
        null
        );

    Session.KeepAlive += new KeepAliveEventHandler(Notification_KeepAlive);
}

I saw in #2255 you have change something but I'don't know why my code don't work in new version. 

Someone can help me? 
Thanks

### Environment

```markdown
- OS:Windows 10
- Environment:Visual studio 2022 (64bit) 17.7.6
- Runtime:
- Nuget Version: 1.4.371.96 and 1.4.372.76
- Component:
- Server:
- Client:
```

mregen, 2023-11-10:
Session.Create is now fully async, at first I'd suggest to add .ConfigureAwait(false) to all awaited calls. It could be the root cause to end up in a dead lock. Please let me know if that makes a difference.

NewMabo, 2023-11-10:
Hi,
Thank you for your quick response.
That make a difference now my code work again.
Thanks

closed 2023-11-11 by mregen