OPCFoundation/UA-.NETStandard #2374: ConsoleReferenceClient UAClient reusable after dispose

DvdKhl, 2023-11-10:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

After creating an UAClient instance and disposing it, it is possible to call the ConnectAsync method successfully.

### Expected Behavior

After disposing the client, the connect method should throw the `ObjectDisposedException` Exception.

### Steps To Reproduce

[OPCNecromancer.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/13317541/OPCNecromancer.zip)

Just execute after setting the url and compiling the sample.

Current console output:
```
Connecting to... opc.tcp://192.168.0.2:4840
New Session Created with SessionName = Quickstart Console Reference Client
Disconnecting...
Session Disconnected.
Disposing Client
Connect after dispose should fail
Connecting to... opc.tcp://192.168.0.2:4840
New Session Created with SessionName = Quickstart Console Reference Client
Disconnecting...
Session Disconnected.
But doesn't
```

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022
- Runtime: .NET 6
- Nuget Version: 1.4.372.76
- Component: OPCFoundation.NetStandard.Opc.Ua
- Server: -
- Client: ConsoleReferenceClient
```

EthanChangAED, 2023-11-17:
Hello @DvdKhl, thank you for reporting this. When you say "possible" what is the rough possibility/frequency of this symptom?  Is it a delay of disposing or the connection can be successful no matter how long after the disoising?

mregen, 2023-11-18:
This is a bug in the usage pattern of the UA Client sample, lets fix it...

PR #2601 (2024-04-27): fix disposal of ConsoleReferenceClient
## Proposed changes

Make ConsoleReferenceClient throw an object disposed exception when trying to use after dispose

## Related Issues

- Closes #2374

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected, requires version increase of Nuget packages)
- [ ] Documentation Update (if none of the othe [... cut]
closed 2024-04-27 by mregen