OPCFoundation/UA-.NETStandard #2390: <title>Start reading node values when the item becomes available

rafalsli, 2023-11-21:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When creating subscription to tags, some of which are not created on the server yet, the client starts reading values from the items which are available. For non existing tags `MonitoredItem` objects have `Status` with `Error` `BadNodeIdUnknown`. This is expected behaviour. 

However, when tag becomes available status of the `MonitoredItem` is not changed and client does not read values of newly created item.

### Expected Behavior

I think that when node starts to be available status of `MonitoredItem` should change and client should start receiving data.

Is there any way at this point to observe the real status of items which were not available at the moment of adding them to subscription and then force the subscription to start receiving values?

### Steps To Reproduce

1. Create subscription with node which is not available on the server
2. Configure missing tag on the server

### Environment

```markdown
- OS: Windows 11
- Environment: 
- Runtime:.NET 6
- Nuget Version: 1.4.363.107
- Component: Opc.Ua.Client
- Server: KEPServer
```

EthanChangAED, 2023-11-28:
Related to #1904

mrsuciu, 2023-12-18:
@rafalsli 

I've tried to create such a "non-existent tag" MonitoredItem with the KEPServer but the returned service result from the CreateMonotoredItem service call was `BadNodeIdUnknown`. 

I think the behavior which you experience is "normal" even if it does not intuitively seem so, since when the node does not exist on the server side, the server cannot create the corresponding Monitored Item on the server side also and handle its state.
The MonitoredItem on the server side and the actual Node (tag) it monitors are decupled in the sense that if the node disappears from the address space, than the state of the MonitoredItem can reflect it in its status,  but such a MonitoredItem must exist though.  

The behavior you would like to exist on the client side Monitored Item implies that the Server is also capable to handle such a functionality and must "collaborate" in doing so with the client. 
That would suppose a mechanism of some sort to signal the client that a node is created, via an event most likely (AuditAddNodesEventType exists for auditing purpose but shouldn't count since only privileged client users can consume them). Then the client being subscribed to such an event can create the Monitored Item.

closed 2024-03-20 by EthanChangAED