OPCFoundation/UA-.NETStandard #2394: BadTooManySessions error seen when active session are at around 30 while max session count configured is 100.

vinaybr, 2023-11-24:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

BadTooManySessions error seen when active session are at around 30 while max session count configured is 100. We are using UAExpert as a client. 

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

vinaybr, 2024-11-13:
@mregen do you have any updates on this issue.

romanett, 2025-01-15:
@vinaybr can you provide some more information about the issues you are experiencing

closed 2025-12-03 by romanett