OPCFoundation/UA-.NETStandard #2396: Question: Documentation on profiles

Razuox, 2023-11-27:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Is there some kind of documentation on which [OPC Profiles](https://profiles.opcfoundation.org/profile/?pg=3&rs=3) this SDK implements?

EthanChangAED, 2023-11-28:
@Razuox , thank you for the feedback. We currently don't have good documentation for this area, but will put this item in our backlog.

Razuox, 2023-11-29:
> @Razuox , thank you for the feedback. We currently don't have good documentation for this area, but will put this item in our backlog.

@EthanChangAED Thanks for your reply, so i guess the only information on functionalities without testing the SDK are the [features included](https://github.com/OPCFoundation/UA-.NETStandard#features-included)?

Thanks in advance!