OPCFoundation/UA-.NETStandard #2397: Using a certificate created by New-SelfSignedCertificate

dekenless, 2023-11-27:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

Is there an example of how one would create a certificate using the [powershell routine](https://learn.microsoft.com/en-us/powershell/module/pki/new-selfsignedcertificate?view=windowsserver2022-ps) (New-SelfSignedCertificate) so it can be referenced via the ApplicationInstance? All my attempts have failed.

### Environment

```markdown
- OS: Windows
- Environment:
- Runtime: .NET Standard 2.0
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2023-11-28:
OPC UA needs the SubjectAlternativeName Extension in the X509v3 Cert, if you can figure out how to get this into the created certificate I am confident it will work.
For testing you can also create the Certificate using XCA which builds on the OpenSSL CLI:
https://github.com/chris2511/xca

Siemens created an explanation to modifiy an exisiting cert whose private key you have to include a SubjectAlternativeName  here:
https://support.industry.siemens.com/cs/de/en/view/109812756

dekenless, 2023-12-05:
> OPC UA needs the SubjectAlternativeName Extension in the X509v3 Cert, if you can figure out how to get this into the created certificate I am confident it will work. For testing you can also create [...]

Ok. Thanks, I will have to investigate this further. 

So, as far as you know there is not an official document detailing the certificate requirements?

romanett, 2023-12-05:
The Opc UA Spec Part 2 gives a detailed Overview of the Security Architecute.

dekenless, 2023-12-05:
> The Opc UA Spec Part 2 gives a detailed Overview of the Security Architecute.

Thanks.