OPCFoundation/UA-.NETStandard #2412: Ua Client read Chinese Nodev datavalue will show like this ????25.

acer1204, 2023-12-05:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

use UA-.NETStandard sample client to read siemens opc ua server nodeid with datavalue is chinese word will show like this.
????25.
the scoure text is 刀補號25.

### Expected Behavior

the scoure text is 刀補號25.

### Steps To Reproduce

1.connect to siemens opc ua server
2.read chinese wrod datavalue nodeid.

### Environment

```markdown
- OS:win10
- Environment:vs2017
- Runtime:.net 4.6.2
- Nuget Version:1.4
- Component:Opc.Ua.client
- Server:simemens opc ua server
- Client: softing opc ua client
```

mregen, 2023-12-06:
Hi @acer1204, this is probably a conversion that happens when the Variant value is internally converted to the displayed string, to avoid that non ascii characters are output. Could you provide some more information how the value is printed in your code?

EthanChangAED, 2024-01-02:
closing for lacking of feedback. Feel free to reopen if there is further info/update.

closed 2024-01-02 by EthanChangAED

JoshueC, 2024-10-31:
I have the same problem while reading tool names from siemens 840dsl.

The actual Tool name is "立铣刀“, but I got "???". If the actual Tool name is "T2234", the output is normal.
Get the node from
`var nodeId = new NodeId("/Tool/Catalogue/toolIdent[u1, 1]",  2)`
Get the value from
`var dataValue = session.ReadValue(nodeId)`
Return the result by
`return dataValue.Value`