OPCFoundation/UA-.NETStandard #2417: PublishErrorEventHandler event is not invoked from OnPublishComplete when session reconnect is in progress

vallen-ptc, 2023-12-08:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

An OPC UA client (upstream client) is connected to an aggregation server that has an active client session to a downstream server running in a VM. The upstream client has a subscription with one monitored item from the downstream server showing a status code of "Good".

When the VM with the downstream server is paused the status code for the monitored item in the upstream client remains "Good" indefinitely.

### Expected Behavior

When the VM with the downstream server is paused the status code for the monitored item in the upstream client should transition to "BadOutOfService".

### Steps To Reproduce

1. Connect an OPC UA client (upstream client) to an aggregation server that has a downstream connection to an OPC UA server  running in a VM (downstream server).
2. In the upstream client browse to a node in the downstream server and add a monitored item for the node.
3. Pause the VM where the downstream server is running
4. Wait a minute or so and notice that the status code for the monitored item in the upstream client remains "Good" indefinitely

### Environment

```markdown
- OS: Windows 11
- Environment: N/A
- Runtime: .NET 6.0.10
- Nuget Version: 1.4.371.86
- Component: Opc.Ua.Client.Session
- Server: Quick Start Reference Server
- Client: UA Expert
```

### Anything else?

Technical Description of Issue:

In the OnPublishComplete method (Opc.Ua.Client\Session.cs, line 5102), processing of the publish response throws an exception causing the catch logic to be executed. In the catch logic prior to the code that invokes the PublishErrorEventHandler callback there is a check for a reconnect in progress. If there is a reconnect in progress the method exits immediately skipping the invocation of the PublishErrorEventHandler callback which results in the monitored item status for the upstream client session never transitioning to "BadOutOfService".

Question: 

In the event of an error when processing the publish response why is the PublishErrorEventHandler callback not invoked if the session is in a reconnecting state? Is it possible to always invoke the PublishErrorEventHandler callback regardless of the reconnecting state? 

Our implementation uses the PublishErrorEventHandler event to update the status of the monitored item in the upstream client's session so that the upstream client can show the monitored item quality is bad when a previously existing downstream connection is in a temporary error state".

closed 2023-12-29 by mregen

mregen, 2023-12-29:
@vallen-ptc   please test fix with 1.4.372.112-preview

vallen-ptc, 2024-01-17:
Hi @mregen,

This is my first time testing a stack fix. Does producing the release for version 1.4.372.112-preview in github also produce a NuGet package that I can reference/build our application against or do I need to checkout out the stack sources based in the release commit and build the stack locally?

mregen, 2024-01-17:
Hi @vallen-ptc, there are multiple options to test a fix:
i) check out source code and build with your app
ii) check out source code and produce private Nuget packages which you can consume using a modified `Nuget.Config`
iii) every commit in master is available for some time in the preview channel: https://opcfoundation.visualstudio.com/opcua-netstandard/_artifacts/feed/opcua-preview/connect
iv) in this specific case the preview Nuget package was also released on Nuget.org, so you can just test with it.

mregen, 2024-01-24:
Hi @vallen-ptc, I *think* its not the root cause from what you describe, but 112-preview had a regression which caused subscription publish responses not being acknowledged. Some servers don't like that.
116-preview had been the best option to test against.
Otherwise its hard to tell. Processing of keep alive has become non blocking, along with some other optimizations that may have side effects. So I guess its the way the aggregation server / client interact.