OPCFoundation/UA-.NETStandard #2446: NodeState.RemoveReference may remove the wrong reference

einarmo, 2024-01-03:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

`NodeState.RemoveReference` now removes the _first_ reference that happens to have the requested `TargetId`, which will often be incorrect. This was changed in #2345, but the fix does not make sense to me.

### Expected Behavior

`RemoveReference` is a public method that is stated to remove the reference defined by `(referenceTypeId, isInverse, targetId)`, which it currently _does not do_. It also incorrectly reports which reference it _did_ remove.

### Steps To Reproduce

```
myState.AddReference(ReferenceTypeIds.Organizes, false, nodeId);
myState.AddReference(ReferenceTypeIds.GeneratesEvent, false, nodeId);
myState.RemoveReference(ReferenceTypeIds.GeneratesEvent, false, nodeId);
// Throws an exception, due to a conflict.
myState.AddReference(ReferenceTypeIds.GeneratesEvent, false, nodeId);
```

### Environment

```markdown
- OS: Linux, Fedora
- Environment: CLI
- Runtime: .NET 8
- Nuget Version: 6.8.0.32767
- Component: Opc.Ua.Core
- Server: My own server, but this should be reproducible in the reference server
- Client: None
```

mregen, 2024-01-04:
Hi @einarmo, please check the modified fix and let me know if it is back working.

einarmo, 2024-01-04:
This fixes my issue, but it still seems like weird behavior for `RemoveReference` to do this. It is very unintuitive that it can in some cases remove a _different_ reference from the one you asked to remove, but I don't quite know what #2343 actually is.

mregen, 2024-01-04:
I think your opinion is correct. I double checked and it looks like the error happens in another layer. Once verified, we go back to the original code in NodeState. Thanks for checking!

mregen, 2024-01-12:
The regression qualifies for a hotfix for .106

mregen, 2024-01-16:
see 1.4.372.107

PR #2450 (2024-01-05): Fix nodestate remove for diagnostic infos
## Proposed changes

Issue:
- In order to fix #2343, the change caused a regression pointed out by #2446.

Fix:

- First try to remove node, only then try to remove the reference to the target node.

Repro:
- connect/disconnect to the server with diagnostics enabled. Check Diagnostic nodes for leaking references.

## Related Issues

- Fixes #2446 , #2343

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which f [... cut]
PR #2456 (2024-01-15): Release hotfix: Fix nodestate remove for diagnostic infos (#2450)
- revert fix for regression introduced by fix for #2343, apply fix in nodemanagers which do not invert the inverse property, so bidirectional references were not removed.

## Proposed changes

Describe the changes here to communicate to the maintainers why we should accept this pull request. If it fixes a bug or resolves a feature request, be sure to link to that issue.

## Related Issues

- Fixes #2446 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill t [... cut]
closed 2024-01-16 by mregen