OPCFoundation/UA-.NETStandard #2469: Make UA-.NETStandard GDS-Feature Complete

romanett, 2024-01-19:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [X] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

Some Important features are currently misssing in the UA-.NET GDS implementation and related libraries.
Some are already adressed and some will be possible to implement, now that 1.05.03 nodeset is merged:
- Application Self Admin privilege: #2338  (Last tests are currently being implemented
- Use CA signed Cert for GDS: #2375 The Tests from #2338 will help with implementing the Certificate exchange
- Make the Client Support Pull Model Authorization. The Tests from #2338 will also set the foundation here to exchange the Application Instance Cert at runtime and the necesary method calls.

New Features from 1.05.03. Nodeset release #2423:

- Methods for Certificate Directory Type:
https://github.com/OPCFoundation/UA-.NETStandard/blob/7867a6fa7ea6dd2e0fe4e04d86d7e3cc6c358ddd/Stack/Opc.Ua.Core/Stack/Generated/Opc.Ua.Gds.Constants.cs#L207C1-L214C128

        public const uint CertificateDirectoryType_CertificateGroups_DefaultApplicationGroup_TrustList_CloseAndUpdate = 538;

        public const uint CertificateDirectoryType_CertificateGroups_DefaultApplicationGroup_TrustList_AddCertificate = 541;

        public const uint CertificateDirectoryType_CertificateGroups_DefaultApplicationGroup_TrustList_RemoveCertificate = 543;

Node Ids: Line 3958.. , 4030..., 4102..., 

(Same for DefaultHttpsGroup & DefaultUserTokenGroup)
Also for the CertificateDirectoryType itself:
https://github.com/OPCFoundation/UA-.NETStandard/blob/7867a6fa7ea6dd2e0fe4e04d86d7e3cc6c358ddd/Stack/Opc.Ua.Core/Stack/Generated/Opc.Ua.Gds.Constants.cs#L418C9-L427C80
       public const uint CertificateDirectoryType_GetCertificates = 89;

        public const uint CertificateDirectoryType_GetTrustList = 197;

        public const uint CertificateDirectoryType_GetCertificateStatus = 222;

        public const uint CertificateDirectoryType_CheckRevocationStatus = 126;

NodeIds Line: 4169...

- WellKnownRoles for GDS:
https://github.com/OPCFoundation/UA-.NETStandard/blob/7867a6fa7ea6dd2e0fe4e04d86d7e3cc6c358ddd/Stack/Opc.Ua.Core/Stack/Generated/Opc.Ua.Gds.Constants.cs#L719C7-L732C74
  
        public const uint WellKnownRole_DiscoveryAdmin = 1661;

        public const uint WellKnownRole_CertificateAuthorityAdmin = 1680;

        public const uint WellKnownRole_RegistrationAuthorityAdmin = 1699;

        public const uint WellKnownRole_KeyCredentialAdmin = 1718;

        public const uint WellKnownRole_AuthorizationServiceAdmin = 1737;

NodeIds: Lines 4470..

Browse Names: Line 7704-7716

### Expected Behavior

Please help me make the GDS Feature complete.
I am open for collaboration on new features and pull requests

romanett, 2024-02-13:
Well Known Roles Matrix for GDS

Method |   | DiscoveryAdmin | CertificateAuthorityAdmin | RegistrationAuthorityAdmin | SecurityAdmin | ApplicationSelfAdmin | ApplicationAdmin |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  |   |   |   |   |   |   |   |   |   | Combinations |   |   |  
DirectoryType |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   | CAAdmin + Securiy Admin |   |   |  
RegisterApplication |   | x |   |   |   |   | x |   |   |   |   |   |  
UpdateApplication |   | x |   |   |   | x | x |   |   | DiscoveryAdmin + ApplicationAdmin |   |   |  
UnregisterApplication |   | x |   |   |   | x | x |   |   |   |   |   |  
GetApplication |   |   |   |   |   |   |   |   |   |   |   |   |  
FindApplications |   |   |   |   |   |   |   |   |   | DiscoveryAdmin + Application(self)Admin |   |   |  
QueryApplications |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
TrustListType |   |   |   |   |   |   |   |   |   | CAAdmin + Security Admin + Application(self)Admin |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
OpenWithMasks |   |   | x |   | x | x | x |   |   |   |   |   |  
CloseAndUpdate |   |   |   |   |   |   |   |   |   | SecurityAdmin |   |   |  
AddCertificate |   |   | x |   | x |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
CertificateGroupType |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   | CAAdmin |   |   |  
GetRejectedList |   |   |   |   | x |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   | CAAdmin  + Application(self)Admin |   |   |  
ServerConfigurationType |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
UpdateCertificate |   |   |   |   | x |   |   |   |   |   |   |   |  
GetCertificates |   |   |   |   | x |   |   |   |   |   |   |   |  
Create SigningRequest |   |   |   |   | x |   |   |   |   |   |   |   |  
ResetToServerDefaults |   |   |   |   | x |   |   |   |   |   |   |   |  
Apply Changes |   |   |   |   | x |   |   |   | from the Session that created the transaction |   |   |   |  
CancelChanges |   |   |   |   | x |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
CertificateDirectoryType |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
StartSigningRequest |   |   | x |   |   | x | x |   |   |   |   |   |  
Start NewKeyPairRequest |   |   | x |   |   | x | x |   |   |   |   |   |  
FinishRequest |   |   | x |   |   | x | x |   |   |   |   |   |  
RevokeCertificate |   |   | x |   |   |   |   |   |   |   |   |   |  
GetCertificateGroups |   |   | x |   |   | x | x |   |   |   |   |   |  
GetCertificates |   |   | x |   |   | x | x |   |   |   |   |   |  
GetTrustList |   |   | x |   |   | x | x |   |   |   |   |   |  
GetCertificateStatus |   |   | x |   |   | x | x |   |   |   |   |   |  
Check Revocation Status |   |   |   |   |   |   |   |   | from an authenticated SecureChannel |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   | can Approve Requests |   |   |   |   |   |   |   |   |

mregen, 2024-02-27:
@romanett please update the issue after the Feb release which work items are still msissing.

PR #2554 (2024-03-29): GDS: Add Audit Events to GDS and missing Events to ServerConfigurationNode
## Proposed changes

Make the GDS an Auditing Server by adding all needed Audit Events to the GDS:

ApplicationRegistrationChangedAuditEventType
https://reference.opcfoundation.org/GDS/v105/docs/6.6.12

CertificateDeliveredAuditEventType
https://reference.opcfoundation.org/GDS/v105/docs/7.9.13

CertificateRequestedAuditEventType
https://reference.opcfoundation.org/GDS/v105/docs/7.9.12

Add missing evetns to Server ConfigurationNode:

TrustListUpdateRequestedAuditEventType
https://reference.opcfoundation.org/GDS/v105/doc [... cut]
PR #2553 (2024-06-05): [GDS] Add Method GetCertificates to GDS for Pull Support and ServerConfiguration for Push Support / Fix CheckRevocationStatus
## Proposed changes

Adding the Method GetCertificates for Pull Model to the GDS:
https://reference.opcfoundation.org/GDS/v105/docs/7.9.8

And also to the Server for Push Support:

https://reference.opcfoundation.org/GDS/v105/docs/7.10.5

Additionally the GDS Client & Push client are updated with the new Method

Fix checkRevocationStatus Method to do an online Revocation check.

## Related Issues

- Fixes #2469

## Types of changes

- [ ] Bugfix (non-breaking change whi [... cut]
closed 2024-03-13 by romanett