OPCFoundation/UA-.NETStandard #2471: RevokeCertificate API not yet in SDK?

charlesaugustineabb, 2024-01-22:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

RevokeCertificate API is defined in spec12 1.05.02 . however probably not implemented in SDK. when it is planned to include in SDK?

### Expected Behavior

RevokeCertificate API is defined in spec12 1.05.02 . however probably not implemented in SDK. when it is planned to include in SDK?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-01-22:
We can implement it together. If you create a pull request/frok branch. I am happy to collaborate

romanett, 2024-02-01:
@mregen How do I activate the Method, it appears in the structure, but to me it seems the RevokeCertificateMethodStateMethodCallHandler is not activated:

[image]

[image]

mregen, 2024-02-27:
@charlesaugustineabb please check the Feb release for updates regarding the GDS.

PR #2497 (2024-02-29): GDS: add Method RevokeCertificate to Client and Server
## Proposed changes

Implement the method RefokeCertificate from Spec: [7.9.6 RevokeCertificate (https://reference.opcfoundation.org/GDS/v105/docs/7.9.6)]

## Related Issues

- Fixes #2471

## Types of changes

- [ ] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected, requires version increase of Nuget pack [... cut]
closed 2024-02-29 by mregen