OPCFoundation/UA-.NETStandard #2473: Adding RevokeReason parameter for RevokeCertificate API

charlesaugustineabb, 2024-01-23:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

RevokeCertificate (
[in] NodeId applicationId
[in] ByteString certificate); 

### Expected Behavior

RevokeCertificate (
[in] NodeId applicationId
[in] ByteString certificate
[in] String revokereason);

any possibility to add revokereason parameter to RevokeCertificate  API.
example value for reasons are  as below 
1)Unspecified
2)Key Compromise
3)Suspended
4)Cessation of operation
etc...

datatype of  revokereason can be  enum  instead of string.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-01-23:
see 

https://github.com/OPCFoundation/UA-.NETStandard/blob/7867a6fa7ea6dd2e0fe4e04d86d7e3cc6c358ddd/Libraries/Opc.Ua.Gds.Server.Common/ApplicationsNodeManager.cs#L421

https://github.com/OPCFoundation/UA-.NETStandard/blob/7867a6fa7ea6dd2e0fe4e04d86d7e3cc6c358ddd/Libraries/Opc.Ua.Gds.Server.Common/ApplicationsNodeManager.cs#L273

As per 1.05.02 spec this is not in the standard:

https://reference.opcfoundation.org/nodesets/6/17373

For you own fork you could add this however and be not conform with the standard

charlesaugustineabb, 2024-01-24:
I would like to make a request add it to the specification . that was my intention(made the issue as enhancement). what is the process for same, could you please advise me?

mregen, 2024-01-26:
Hi @charlesaugustineabb , the process is to open a Mantis issue here: https://mantis.opcfoundation.org/, select Part12 - 10000-012 Discovery.
Then the working group will pick it up and discuss.  
As OPCF member you should be able to get access.

charlesaugustineabb, 2024-02-01:
I am not able to log in https://mantis.opcfoundation.org/
can i get permission to log in above link , so i can log in and submit my enhancement request

charlesaugustineabb, 2024-02-29:
Still i am not able to submit a case in mantis.

EthanChangAED, 2024-03-20:
This is more of a request to the spec. Closing as there is not action item for the .net stack side. Feel free to reactive if the spec is updated.

closed 2024-03-20 by EthanChangAED