OPCFoundation/UA-.NETStandard #2486: Introduce ISubscription

KircMax, 2024-01-30:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Currently there is no abstraction for the Opc.Ua.Client Subscription class - it implements IDisposable and ICloneable -> 
[image]
As a library user I'd like to have the possibility to implement my own custom Subscription that e.g. has a default Setting for the monitored Item that will be created 

### Expected Behavior

ISubscription exists and is used e.g. in the Session.cs, etc.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

KircMax, 2024-01-30:
Would that be desirable from your side aswell? (Then I might implement a proposal for this)

mregen, 2024-01-31:
Have you tried to subclass Subscription yet? Its now possible by overriding the 
`public virtual Subscription CloneSubscription(bool copyEventHandlers)`.
A similar concept exists for monitored items.
Apparently ISubscription is always better, but in the short run you could use the subclassed versions.

EthanChangAED, 2024-03-20:
Assuming the subclass is a validation short-term solution. Feel free to reactive this issue if there is still needs for the ISubscription.

closed 2024-03-20 by EthanChangAED