OPCFoundation/UA-.NETStandard #2492: A&C Enhancements for the Client

KircMax, 2024-01-31:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Currently it's not that easy to e.g. Confirm, Acknowledge an Event/Alarm, etc.

### Expected Behavior

I think there could be some enhancements to simplify the usage as a library user.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

KircMax, 2024-01-31:
If you think that that would be desirable I could prepare a proposal and ask for your feedback on it.

mregen, 2024-01-31:
Absolutely, please prepare your proposal!

KircMax, 2024-03-01:
Maybe you could point me to some places if I got things wrong/ want to put things in the wrong place/ etc. - which I can very well imagine.
Here a screen of the 'acknowledgeableEvent'
[image]

EthanChangAED, 2024-03-20:
@mregen, would you take a look at KircMax' PR?

romanett, 2026-02-14:
@KircMax please reopen a new PR if you have some general impl ready to be included in the SDK

closed 2026-02-14 by romanett