OPCFoundation/UA-.NETStandard #2517: No Notification on Monitored Item in ConsoleReferenceClient Sample

KircMax, 2024-02-12:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

When I attach to the Monitored Items Notification in the Subscription Sample of the ConsoleReferenceClient I do not hit a breakpoint that I attached to the Eventhandler for the Notification...:
[image]
I thought maybe some other MonitoredItem is used internally and if I use the one after ApplyChanges it might work
but these breakpoints are also never hit
[image]
So I guess I am doing something else wrong...

If I do extend the FastDataChangeCallBack that is provided to the subscription
[image]
[image]

then I can see that the entries are written for the according notifications ... 
I guess I am missing something?
Thanks in advance.

### Expected Behavior

Notification (Eventhandler) is executed before/after the subscription FastDataChangeCallBack is called.

### Steps To Reproduce

Attached the ClientSamples.cs and Program.cs I touched in the ConsoleReferenceClient.
(the node var nId = new NodeId(6107, 4); of course needs to be changed to one (struct) that is present on your server).
[Sample.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/14240963/Sample.zip)

### Environment

```markdown
- OS: Windows 10
- Environment: Visual Studio 2022 17.8.5
- Runtime: .NET 6.0 
- Nuget Version: 
- Component: ConsoleReferenceClient
- Server: S7-1500
- Client: ConsoleReferenceClient
```

mregen, 2024-02-14:
Hi @KircMax, I *think* this might be because of `DisableMonitoredItemCache`, then the callbacks in `SaveValueInCache` are not executed.

KircMax, 2024-02-16:
Hi @mregen yes - that was it, thanks a lot

closed 2024-02-16 by KircMax