OPCFoundation/UA-.NETStandard #2525: The PolicyId that has been used to activate the session is not stored for reconnect ("reactivatesession")

KircMax, 2024-02-16:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When using a userIdentity with a specific PolicyId for the connect:
[image]
The PolicyId will not be re-used for the reconnect:
[image]
But the OPC Spec. says that the Same PolicyId must be used to re-authenticate that user.
[image]
https://reference.opcfoundation.org/Core/Part4/v105/docs/5.6.3

### Expected Behavior

the same usertokenpolicy is used for the reconnect that has been used for the connect

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2024-02-16:
Thanks for catching this, I think it is rather a bug.

KircMax, 2024-02-16:
Very welcome! 
Yeah, right, current behaviour does not fit to the OPC spec. -> relabeled.

PR #2526 (2024-02-23): reuse the identity policy in the reconnect that has been used in the connect
## Proposed changes

Describe the changes here to communicate to the maintainers why we should accept this pull request. If it fixes a bug or resolves a feature request, be sure to link to that issue.

## Related Issues

- Fixes #2525 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ]  [... cut]
closed 2024-02-23 by mregen